headtracker/HeadTracker #4: "TLS Initialization failed"

dlktdr, 2021-02-03:
Issue trying to load firmware. Probably couldn't find the OpenSSL library I packaged with it. Need some more info on the windows version. Maybe I'll switch to a 32bit build for compatibility

kevin-kevjan, 2021-02-06:
in V0.41 and using Win 10 home (below) get "TLS Initialization failed" error when clicking on Upload Firmware menu option.
Step to reproduce: Unzip file to dir on desktop and run HeadTracker.exe. Click "Upload Firmware".

[image]

Edition	Windows 10 Home
Version	20H2
Installed on	‎2/‎4/‎2021
OS build	19042.746
Experience	Windows Feature Experience Pack 120.2212.551.0

jreise-d, 2021-02-10:
Same here. Win10 Pro 64bit Version 20H2. Same on another computer with Version 2004.
I have tested 0.5 and 0.41.
[image]
A strange thing: After the error appears, the file firmwares.ini in the directory HeadTrackerGUI_x64.-.V0.5 is emptied, so it has 0 bytes instead of 2kB

dlktdr, 2021-02-10:
The firmwares.ini file is the one that gets downloaded from the online source. Since it can't download I'm not surprised it's gone.

I'll try to hard link the SSL libraries in the code rather than shipping the .dLL's to see if it makes a difference.

In the meantime can you install http://slproweb.com/products/Win32OpenSSL.html
Should be the "Win64 OpenSSL v1.1.1i Light package"

jreise-d, 2021-02-10:
thank you so much! With the 32bit version it works

dlktdr, 2021-02-11:
Please check out the 0.7 release. I'm going to go to a 32 bit from here on out.

dlktdr, 2021-02-11:
I've found the root cause of this. OpenSSL needs the Microsoft Visual C++ redistributable 2015, which many computers already have installed but not all. 

Please install https://www.microsoft.com/en-ca/download/details.aspx?id=48145

I've included the missing file in the 0.71 release file https://github.com/dlktdr/HeadTracker/blob/master/gui/bin/HeadTrackerGUI_x64%20-%20V0.71.zip

Also cancel the 32bit idea. That wasn't a problem.

closed 2021-02-11 by dlktdr