headtracker/HeadTracker #15: No red "calibration sphere" in V0.7

kevin-kevjan, 2021-02-11:
performed calibration, but no visual red 'calibration sphere' for feedback..
After awhile "Gaps" did go low enough "Save" button was made active.

[image]

dlktdr, 2021-02-11:
Well at least it still worked even though the visualization didn't show.

Try to fix a "TLS initialization" from missing SSL libs and now probably can't find the GLUT .dll now.

I'll look into it.

dlktdr, 2021-02-11:
Please try again on 0.71.

kevin-kevjan, 2021-02-13:
Just tried 0.71. Same issue of no mag visualization.

(I no longer get TLS error however after loading VS Redistributable)

[image]

dlktdr, 2021-02-13:
Could you check if you have the following files in your windows folder?
C:\windows\system32\glu32.dll
C:\windows\system32\opengl32.dll

kevin-kevjan, 2021-02-13:
Yes. I do have both those files in my Win\sys32 directory. (one noted below)

Also, I checked on my other laptop (that never had the TLS error) and the red visualization does work correctly on that system.....so it is dependent on this laptop. 

[image]

dlktdr, 2021-02-13:
Ok that's good those are supposed to come with Windows.

Try that google drive link above if you can, and cross your fingers :)

kevin-kevjan, 2021-02-13:
Success!  ....**Yes** the version 0.71_custom did show the visualization.

[image]

dlktdr, 2021-02-13:
Awesome.. Thanks again for testing all these.. I'll make this change permanent in the new ones.

kevin-kevjan, 2021-02-13:
Happy to help…..This is a very cool project and perfect timing for me and several friends looking for easy FPV head trackers.

dlktdr, 2021-02-14:
Released 0.72 Gui with fix

closed 2021-02-14 by dlktdr