headtracker/HeadTracker #16: Version listed in Firware update.

kevin-kevjan, 2021-02-14:
Just downloaded and installed V0.72 using Win10 system that had prior TLS error and no calibration visualization

**ALL GOOD now.** 👍 

However, firmware download list did not show V.72 (although GUI did) but all worked as expected for V0.72

[image]

dlktdr, 2021-02-15:
My plan was if you have a .7x firmware it will work with any .7x GUI. Any minor fixes to either the GUI or firmware version would go up by .01, so no I didn't post a 0.72 firmware, 0.71 is the newest. I guess I can see how this could be confusing.

Maybe I'll switch the numbering scheme to be completely different. then just grab the newest available.

I think it's getting stable enough I can call it 1.0 pretty soon, maybe I'll switch it up then.

kevin-kevjan, 2021-02-15:
Good to go….minimal confusion and will simply resolve itself as next version is released.

Working is working!

closed 2021-02-15 by kevin-kevjan