headtracker/HeadTracker #22: FrSky horus bluetooth module will not stream data wirelessly, installed in Jumper TX16S

dlktdr, 2021-03-18:
The stand-alone bluetooth module available for the X10 X10S and X12S will not stream data. There is a few variations in the BLE protocol and will not send data to the head tracker.

Two users have this module in a Jumper TX16s and would not function. I would have to assume that it won't work in the X10, X10S and X12S as well.

I have ordered a module to diagnose and see the issue. I believe this stand alone bluetooth would be very helpful to some, so I will spend some time on it after it arrives from China.

SilentZen, 2021-03-19:
I have para wireless in my T16 (not t16s) and it works perfectly.
Take care to note which wireless para module does not work as even on frsky radio's the one with the antenna port (usually branded bluetooth module does not work for trainer function but the one without antenna port (usually branded para wireless module) does.

dlktdr, 2021-03-19:
Thank you. Good to know which ones will work.

Riffer, 2021-03-19:
From my account, at HorusRC I see that mine seems not to be able to PARA - but from your docs @dlktdt, I took, that it should? Will there be a version that will work without PARA? From RCGroups you said, you get the 10 dollar module?

I wonder why the 10 dollar module exits - I can not see Telemetrie data from there and it may be a problem to connect the trainer. What kind of a replacement should this be...

dlktdr, 2021-03-19:
Since I already ordered the 10 dollar (Bluetooth module) before I knew this, I'm going to see if I can make it work with the  Tracker at least. It is streaming data so out so I'm sure there is a way to make it read the data streaming in. Will still give it a go when it arrives

uranus36, 2021-03-19:
Hi Dlktdr,
    My configuration is Radiomaster TX16S with the 10usd bluetooth module. Now I know why I can't establish the PARA communication..... Hope you can come out a solution to the bluetooth module one. Thank you !!

JasFPV, 2021-03-25:
Also have this same issue, Fitted the BT module not realizing that there are two variants. It is connected to the Arduino but not moving any of the channels. Interested to see if there is any development on this. Definitely buying this guy a beer if it works since it will save me buying a new BT module :D

dlktdr, 2021-03-25:
Where is everyone finding out about this product, should I put a disclaimer right on the main Wiki? Or are you finding it elsewhere? I edited my post and in the PARA wiki page and RCgroups as soon as I found out so hopefully no one else will purchase it and need to be posting here too.

Mine shipped from CN on Monday but probably won't be here for two more weeks. I'll put the BLE sniffer on it to see what it's doing and if it's possible to make it work.

JasFPV, 2021-03-26:
> Where is everyone finding out about this product, should I put a disclaimer right on the main Wiki? Or are you finding it elsewhere? I edited my post and in the PARA wiki page and RCgroups as soon [...]

Honestly i cannot remember where i found it but of course in true internet style i got all excited, found the only place in the UK that stocked "X10 bluetooth units" purchased one, fitted it and failed :) Im holding on buying the replacement just incase you work out a way around this bug. All I can say is thank you :)

Riffer, 2021-03-27:
@dlktdr It was mentioned a year ago for the Jumper T16 and I bought the module straight away hoping for benefit when I ordered my Radiomaster. That time I was not aware (and I doubt anyone else) that FrSky exchanges the BT later while upgrading their systems from ACCST to ACCESS.

A disclaimer will help a lot of people avoid the trouble!

For me, if the adaption of the 10$ fails, I will stay with another Arduino in the JR port instead of getting the mess of desoldering the cheap, get the new one for about 30$ plus taxes to find out that other things do not work, too. 

Just now my 3D printer iterates JR modules for the fitting of the secondary Arduino using SBus instead. This will be a good solution and offers the ability to use the Headtracker with the exchangeable modules in my additional TX12.

When my JR box is ready for the masses I will add it here. ;-)

SilentZen, 2021-03-27:
BTW I can confirm not just T16 but also T16s  works with "FrSky ACCESS PARA Wireless Module"
Thread can probably be closed now.

dlktdr, 2021-03-27:
Does anyone know if the trainer works between two of the 10 dollar modules?

SilentZen, 2021-03-27:
It may, also it may work it master/slave is switched.

Riffer, 2021-03-28:
If this can be solved it would also add great value to the OpenTX world. 

So - I am eagerly waiting for your results - I planned to spy the internal serial transmission between the TX and the 10$BT module to see what is going wrong but I have seen your equipment is much better - you own an Oszilloskope that can decipher the transmissions directly. I really wonder what the problem there is - might be a very specific 'modification' of FrSky to the protocol as they already did in the PARA module. I had some time to check the OpenTX source in this regard - and they already worked around some quirks.

rotorman, 2021-04-05:
I did some experiments with FrSky BT module with the antenna socket and posted them in RCGroups:
[https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page662#post46878713](https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page662#post46878713)

Maybe this interesting (especially the end of the post), as I do get the trainer data out from it (the 0x7E 0x80 ... 0x7E messages).

dlktdr, 2021-04-05:
Interesting. Good work on the internal details. 

I see they have laser etched the chip name off my PARA module. I'm guessing it's a CC2540/2541 as well. Curious if they set the code protect bit. Once I get my loose one I'm going to see if I can download the code. I'm guessing they did, would be pretty bad engineering if they forgot on a proprietary system. Could just para code to new to $10 module. Assuming the wiring is the same, which probably is.

I'm hoping mine are going to arrive soon, but they still haven't made it to Canada. Have the BLE sniffer+ Wireshark here ready to see what it's doing on the BLE level. Since OpenTX talks to both modules the same it must be looking for something on the Bluetooth level, or they just never put code on the non-para CC2540 chip to receive 👎. Designed for Telemetery output only?

Will see.

rotorman, 2021-04-05:
Hi Cliff,
this post of mine here might be interesting for the flashing of the BT module chip:
[https://www.rcgroups.com/forums/showthread.php?3395177-Official-OpenTX-version-2-3-Discussion-Thread/page317#post46796919](https://www.rcgroups.com/forums/showthread.php?3395177-Official-OpenTX-version-2-3-Discussion-Thread/page317#post46796919)
Have not tested this myself, but looking at the binary content, it at least perfectly matches the size of CC2540 flash size.
Best,
Risto

rotorman, 2021-04-07:
I made a quick test HeadTracker v0.91 fw, the cheap FrSky Bluetooth module (with external antenna), while serial capture lead was additionally connected to BT module TX pin (=TX16S UART6 RX pin).

The HeadTracker GUI (v0.90) shows green light on Bluetooth connected status, also OpenTX on TX16S finds the Arduino Nano 33 BLE Sense (Bluetooth address of Nano 33 BLE Sensoe shows up under MODEL SETUP -> Trainer -> Master/Bluetooth).

The serial trace only shows "Connected:7F1C17xxxxxx", but no further data gets apparently sent.
After powering down Nano 33 BLE Sense, after some seconds, I see "DisConnected" and repeated "OK+CONNA" messages every 2 seconds.

dlktdr, 2021-04-07:
Starting to wonder if it doesn't have any receive code on the cc2540. Was looking up the development set to write your own programs for it.. ouch... certainly not made for open source. ~$2000 for IAR ide, + debugger probe. :( or limited 30day test :(

@rotorman Do you have the pinout of the BLE module made? & where they connect on the chip. Here is the HM10,
[image]

From looking at your results and what I see in Opentx it looks like the same AT command set at the HM10, similar chip 2541 w/usb vs 2541, same 256 flash size. Like they took that code, changed the BLE service UUID from 0xFFEO to 0xFFF0 +a few other things too so the two won't connect and uploaded it again. Looks like the source is closed on the HM10 one too :( There are a few HM10 clones out there might be some source code out there too.

If the pin out for the TX/RX UART pins (1 + 2) is programmed the same as this HM10, might be able to use this firmware, it's easy to get. I could just check for either service UUID characteristic and bobs your uncle.

https://circuitdigest.com/microcontroller-projects/how-to-flash-the-firmware-on-cloned-hm-10-ble-module-using-arduino-uno
https://github.com/RedBearLab/CCLoader - I see this has some read functionality in it's code too. Haven't looked in more detail. Maybe the para isn't code protected? I'll try..

Just thinking a-text, my modules still not even in Canada yet.

rotorman, 2021-04-07:
In the FrSky module:

- GND - GND
- VCC - +3.3V
- P20 enable (low active) - chip pin 36 P2_0
- P02 UART RX (RX for the BT module) chip pin 17 P0_2
- P03 UART TX (TX for the BT module) chip pin 16 P0_3

[image]
[image]

I do not have a TI CC Debugger, otherwise could try to read it out.

dlktdr, 2021-04-07:
👎  HM10 code won't work. Just mapped out HM10 pins and compared to yours.

UART TX - P1_6 (38)
UART RX - P1_7 (37)

Doesn't look like reading the para code or non-para code wouldn't solve anything either. Pulled my radio apart and the para chip isn't a 2540/1.
[image]
Not sure what it is but it's a 48 pin device.

rotorman, 2021-04-07:
It would be interesting to know if the PARA module is a BLE module at all? Do you have a [BLE sniffer](http://dispert.international-university.eu/ble_files/BLE_sniffer_guide.pdf) to check this?

dlktdr, 2021-04-07:
I do have one, https://www.adafruit.com/product/2269

It comes back as a TI something while it's advertising. 

[image]

rotorman, 2021-04-07:
That limits the search quite nicely. I could find BLE products from TI with 48-pin VQFN package: CC2640/2642/2650/2652.
[https://www.ti.com/wireless-connectivity/bluetooth/products.html#p2954=VQFN](https://www.ti.com/wireless-connectivity/bluetooth/products.html#p2954=VQFN)
Whereas CC2540/41 are 8051-core devices, the 48-pin packages use Cortex-M3 and M4F, thus I guess at least the low-level code will be greatly different.

rotorman, 2021-04-07:
Just had a chat with TI rep, he thinks from the pics and your scan it is CC2650, but promised to come back to me to see if he can match your hw id to a internal db of make/model.

rotorman, 2021-04-07:
DIO_24 and DIO_25 (pins 37 and 38) on CC2650 can be UART:
[https://www.ti.com/document-viewer/CC2650/datasheet/terminal_configuration_and_functions#SWRS1589545](https://www.ti.com/document-viewer/CC2650/datasheet/terminal_configuration_and_functions#SWRS1589545)

SilentZen, 2021-04-07:
I can connect to my tx using BLE Scanner (both in trainer mode and
telemetry)
What info do you need?

BLE Scanner Chart Data
Device Name  Hello
Mac Address  18:04:ED:62:5F:13
Time  RSSI
07-04-2021 20:48:46:083 -64

rotorman, 2021-04-07:
Your ID seems completely from a different batch that Cliffs. Weird. Can you extract somehow what chip it is? Do you have a possibility to attach it to a TI programmer/debugger? These typically verify hw ID by connecting.

dlktdr, 2021-04-07:
Managed to track down some pins on it. Was trying to track down the jtag pins to tie on with my jlink but can't find where TDO and TDI go, unless there are some pads on the bottom for programming, don't really want to unsolder it yet till the others get here as I'm still using it for testing. But as I'm following it through things seem to line up with the cc2650. Crystals in the right place, RF pins match, looks like your right @rotorman.

I know reading the code off this will be useless for fixing the problem at hand.. But If you laser the names off a chips I'm want to know why. Secrets out FRsky. not that we can do anything with that info. I'm sure it's still page protected flash.

[image]

rotorman, 2021-04-08:
I had a wild idea, in order to not have to reverse engineer stuff, to design with proprietary time or size limited GUI: what if we make a custom BLE module for OpenTX radios, matching exactly the pinout (basically it's only 3.3V, GND, Enable, UART RX and UART TX) and use the uBlox Nina-B306 with it's nRF52840 for this.
For PCB Layout could take the Arduino Nano 33 BLE as base and rip it off of everything not needed (which is like 95% of parts on the PCB) and redo this PCB to match FrSky BT size, shape & pinout?
The Eagle files of Arduino Nano 33 BLE are available: https://content.arduino.cc/assets/hw_NANO33BLE.zip
It will be tiny PCB, so even the 4 layer design, which likely might be needed, will be super cheap. Nina-B306 modules are available e.g. by [DigiKey](https://www.digikey.de/products/de/rf-if-and-rfid/rf-transceiver-modules-and-modems/872?k=nina-b306).

It would be easy to implement bi-directional PARA protocol, but also arbitrary better protocols, supporting more channels as well.

dlktdr, 2021-04-08:
Not a bad idea. Would really just be the Ublox on it. Like you said some 4x0.1 pitch pads on one end for SBUS / PPM out + 3.3v reg + little bit of safety circuitry to save when plugged in upside down.

Took a quick look and the BT layout is slightly larger than the Ublox so it would fit... just.

Looked into why you couldn't flash BT from OpenTx yesterday. It's just the header of the .frsk file that needs to be setup properly, so a flash update in radio looks possible. So no USB port required.

Could communicate with just 10bits/ch no escape or start characters since all BLE error checking is done at a lower level + no escape sequence req. but would loose compatibility. Sticking PARA isn't that bad, and other than a few start/stop and odd escape here and there it's pretty well packed. Would just need to change the two hardcoded 8's in OpenTX for more CH's. But then again as a trainer/tracker do we really need more than 8?

Looks like from a quick check here in CAD$ would only cost about $30 to mfg. @ 100qty having a CN company make boards and assemble. Could compete with the price of the PARA modules as long as no intention of making a real profit.

After all this the only thing we would really gain over purchasing a PARA module is the external plug in + not that much cheaper. 

What's everyone else think?

rotorman, 2021-04-08:
Size comparison:
[image]

Very tight...

About the OpenTX BT flashing with FrSky X12S blemodule.bin, I also had further look into OpenTX code. Apparently there needs to be header with [FrSkyFirmwareInformation struct](https://github.com/opentx/opentx/blob/1f593c4f9edbaf4f95c9f7e8c3a0375dfd34f595/radio/src/io/frsky_firmware_update.h#L99-L109), where fourcc = 0x4B535246, headerVersion = 1, productFamily = 4 (=FIRMWARE_FAMILY_BLUETOOTH_CHIP), crc is apparently not checked.

I also see in OpenTX code "CC26xx" strings, the module with ext. antenna I have has CC2540, so I guess the update is for the PARA module.

JasFPV, 2021-04-08:
I desoldered mine and put on the PARA chip and its working fine. Although not quite as smooth as i would have hoped. get a slight jitter if panning at a certain speed any advice on how to get a silky smooth movement on the servos?

dlktdr, 2021-04-08:
> I desoldered mine and put on the PARA chip and its working fine. 

@JasFPV I'm starting to think my modules are lost, still doesn't show they have landed in Canada. You ordered yours after I did and got it installed already :( Glad it works for you tho.

Riffer, 2021-04-09:
@dlktdr, how long did HorusRC take to send them out after order anyways? Mine was about a week ago, and that is still awaiting shipment. I will try to cancel that order. Luckily a seller here in Germany has stock at a reasonable price.

dlktdr, 2021-04-09:
Ordered on the 19th, Shipped on the 23.. I just checked again and it was received in Canada today. Shows 3-7 days for delivery now, but it's already in city about an hour away, so I'm guessing will actually be here quicker.

Riffer, 2021-04-13:
In the meantime, I got one from the german dealer. Now (after exchanging the mainboard and soldering the 'better' (I should say more expensive, right?) module I can confirm that module works - now I am out of helping in this topic. The old mainboard with the 10$ module is on its way to @rotorman.

rotorman, 2021-04-13:
TI finally reported back to me, but not directly with an answer we were looking for, but with a tip how we can easily get to the correct chip type. They wrote:
```
Upon discussing with the product team, it is not currently possible with the provided data (along with the images) to pull-out the device information. I apologize about this. However, if you are able to connect the device via JTAG/a debugger, Uniflash or SmartRF Studio should be able to provide this information.

Uniflash - https://www.ti.com/tool/UNIFLASH
SmartRF Studio - https://www.ti.com/tool/download/SMARTRFTM-STUDIO

```

dlktdr, 2021-04-14:
So my modules showed up today.. the 10 dollar ones.

And good news.. trainer works between two of them. Which means it's something on the Bluetooth level that should be solvable.

Still won't be able to communicate with para to non-para if u you want to use as a trainer but should be able to make the tracker work.

Will sniff what their saying later tonight. And see what's going on but it's nice out so have to go flying first :)

rotorman, 2021-04-14:
Got myself the original [TI CC2540DK-MINI](https://www.ti.com/tool/CC2541DK-MINI) development kit with bunch of components and planning to try to read out the module @Riffer shipped me (will probably arrive any day now). It even came with a nice USB BLE sniffer stick among other parts:

[image]

Need to figure out how to connect the TI CC Debugger to the chip. Any tips?

dlktdr, 2021-04-14:
Awesome.

I used a needle with an alligator clip attached and just ohm out to the pads on the PARA. Looking for the p2.2, p2.1, reset on that one I guess. I don't see any programming pads so the must have brought them out to unused pads on the board.

I found this too,
https://reverseengineering.stackexchange.com/questions/17601/reverse-engineering-8051-firmware
Tried with the cc2540 firmware from the horus and ghidra but couldn't get it to go.. Although I didn't spend too much time. 

A free 8051 compiler that would probably work with the decompiled code.
https://www.instructables.com/8051-Programming-Using-Small-Device-C-Compiler-SDC/

Since I got the two cheaper modules to connect 👍  I have high hopes now I can make it work with the head tracker.. Was having issues with my BLE sniffer have to sort out first didn't actually want to show the data.

Maybe your sniffer is better? I wouldn't suggest anyone purchase the adafruit one..

rotorman, 2021-04-14:
Was reading [CC Debugger user guide](https://www.ti.com/lit/ug/swru197h/swru197h.pdf?ts=1618416203610&ref_url=https%253A%252F%252Fwww.ti.com%252Fsitesearch%252Fdocs%252Funiversalsearch.tsp%253FsearchTerm%253Dswrc212) and it says I only need to connect: GND, VDD, RESET, DD and DC. Looking at [CC2540 datasheet](https://www.ti.com/lit/ds/symlink/cc2540.pdf?ts=1618370883341&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FCC2540%253FkeyMatch%253DCC2540%2526tisearch%253Dsearch-everything%2526usecase%253DGPN), I see that RESET_N is pin 20, DEBUG_ CLK/P2_2 is pin 34 and DEBUG_DATA/P2_1 is pin 35. Lucikly the CC Debugger can also provide 3.3V via it's pin 9, so nothing else needed.

Just tried following setup:
```
**CC Debugger - FrSky BT pin**
pin 1/GND - GND
pin 2/VSense - VCC
pin 3/DC (Debug clock) - P2_2
pin 4/DD (Debug data) - P2_1
pin 5 - not connected
pin 6 - not connected
pin 7 (nReset) - RES
pin 8 - not connected
pin 9/+3.3V supply - also VCC together with pin 2 of CC Debugger
pin 10 - not connected
```

Hooked it up on my FrSky BT and connected CC Debugger via USB to PC. Happy face, as the light on CC Debugger turned green, meaning it can communicate with something it knows.

Fired up [SmartRF Flash Programmer](https://www.ti.com/tool/FLASH-PROGRAMMER) (for CC2540, the old version v1.12.8) and it detects the CC2540. Happy face.

Next hit Read address and get the following screen:

[image]

Bummer. FrSky did lock the chip. Would have been nice if not, but very unlikely.

Nevertheless, it is possible to erase the chip and apply custom code to it. Need the [IAR Embedded Workbench for 8051](https://www.iar.com/products/architectures/iar-embedded-workbench-for-8051/) although for this.

dlktdr, 2021-04-14:
When you're done testing it would be interesting if the code you found for the Horus uploads and runs on there.

When you're done with it of course. Might brick it completely.

I ordered a different sniffer, to anyone reading this if you want a Bluetooth sniffer don't buy the afafruit one, they just use nordics code anyway.. tried to get it to work on the nano but couldn't, since it's same processor. Waste of money but always find these things later..

This one is cheaper and can run v3 of their code and wireshark still. And ble 5.0

https://www.digikey.ca/en/products/detail/nordic-semiconductor-asa/NRF52840-DONGLE/9491124
Sniffer fw and docs are on nordics site.

dlktdr, 2021-04-17:
Update

Well now I know why it isn't working. The CC2540 does a pairing request & sends encrypted. PARA module doesn't.

Para Module
--------------
[image]

Non-Para Module (CC2540)
---------------
[image]

Arduino BLE Library doesn't have support for pairing at this time. Found someone that forked and added pairing support so might still be possible....

rotorman, 2021-04-17:
Great info! I wonder if FrSky would tell us the key used if one points to this nice open source project and explains the reasons why we ask for the key ?
Another option would be to write the code for CC2540 from scratch, for this might need to persuade IAR again to allow their IDE and compiler to be used for open source project (I hope that they might be cooperative).

dlktdr, 2021-04-17:
FYI If you want to use make the Arduino Nano 33 BLE into a Bluetooth scanner I found out a way. You need a debugger / programmer soldered onto the bottom of the board.

1. Using Jflash lite, write this bootloader (from the NRF SDK)
[open_bootloader_usb_mbr_pca10059_debug.zip](https://github.com/dlktdr/HeadTracker/files/6329606/open_bootloader_usb_mbr_pca10059_debug.zip)

1. Open NRF Connect - Programmer (https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Connect-for-desktop)
Connect to the device at the top left, open this firmware and write it. [sniffer_nrf52840dongle_nrf52840_7cc811f.zip](https://github.com/dlktdr/HeadTracker/files/6329633/sniffer_nrf52840dongle_nrf52840_7cc811f.zip)
Mine always shows, 
[image] Even with the error the flash worked.

1.  * Download Python3, Wireshark, and Wireshark plugin all for Version 3
    * Docs (https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_sniffer_ble%2FUG%2Fsniffer_ble%2Fintro.html)
    * Download (https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE/Download#infotabs)

-----------------------

To go back to the orig bootloader here it is, also in the Arduino library framework If platformIO (C:\Users\cblac\.platformio\packages\framework-arduino-nrf52-mbedos\bootloaders\nano33ble)
[Nano33Bootloader.zip](https://github.com/dlktdr/HeadTracker/files/6329629/Nano33Bootloader.zip)

rotorman, 2021-04-18:
With two non-PARA FrSky BT modules and TI CC2540 sniffer stick, I made a first experiment. I soldered one module to RadioMaster TX16S, running vanilla OpenTX v2.3.11 with Bluetooth enabled. I put it in Master/Bluetooth mode (where it receives data).

The second non-PARA FrSky BT module (thanks to @Riffer, who kindly sent me this), I soldered to [FTDI UB232R converter](https://www.ftdichip.com/Support/Documents/DataSheets/Modules/DS_UB232R.pdf) and hooked to PC running [hTerm](https://www.der-hammer.info/pages/terminal.html).

I let the [Texas Instruments SmartRF Packet Sniffer](https://www.ti.com/tool/PACKET-SNIFFER) in BLE mode running and manually emulate second OpenTX radio in Bluetooth/Slave mode from hTerm using the commands learned by [sniffing the serial communication](https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page662#post46878713) and by reading the [OpenTX src](https://github.com/opentx/opentx/blob/2.3/radio/src/bluetooth.cpp).

In short:
```
AT+BAUD4
AT+TXPW0
AT+ROLE0
```

Then initiating a scan from TX16S and connecting to the FrSky module, connected to PC.

This is the sniffer output after connection:

[image]

I then emulated sending a trainer signal, by repeadeadly sending from hTerm:
`7E 80 DB 5D E5 DC 3D D5 B2 59 E5 DC 55 67 37 7E`
which is throttle down, remaining sticks center state (throttle being channel 3). The arrival of the faked trainer input I can see in channel monitor in TX16S - everything is as expected.

This is the sniffer output during such periodic transmission:

[image]

I guess, the Long Term Key that the TI sniffer software used and lists:
`93:c6:ae:6f:31:bd:fc:54:80:02:eb:63:c5:0d:3f:53`
is just something random and possibly not matching the one used by the modules. Or is it?

dlktdr, 2021-04-19:
@rotorman I like that sniffer way more. Way better of a readout! Although I don't know what to make of it. also still new to BLE.

I noticed there is a 000000 code when you pick the telemetry option in open tx.. is it that simple? I've wrote a simple program with the modified Arduino BLE to try to enable pairing support but haven't had much luck as of yet. Was hoping to get it to the point that I have my phone pop up with a pairing request like it does with the CC2540, but no go so far.

Here is the git hub link to the modified source for the Arduino BLE lib I was trying, https://github.com/unknownconstant/ArduinoBLE

When connecting to the CC2540 with my phone and the NRF scanner app and I'm still a little confused why if there is a pairing request and passcode why does it ever let me connect to the service and allow a notify on the characteristic to see the channel data?
Is there some default code that lets you connect? Feel like I need to read up some more on the internal details.

rotorman, 2021-04-19:
> Is there some default code that lets you connect?

Not that I am aware of. Did not have to enter any code in no dialog to get the output in my last post above. All default values.

twistedwings, 2021-04-30:
Have you noticed these ?

[http://mcheli.blogspot.com/2020/08/radiomaster-tx16s-tuning.html](url)
[https://www.youtube.com/watch?v=a3wXHn0cokg](url)
According to this russian guy the old BT module seems to work for telemetry in TX16S.

[https://frsky-forum.de/lexicon/index.php?entry/185-bluetooth-module-im-zusammenspiel/](url)
This is a german compatability list of the different BT modules.
According to this the old module works in trainer mode if the student transmitter (headtracker) has the old module as well.

SilentZen, 2021-04-30:
Telemetry works fine on old module trainer does not.

twistedwings, 2021-04-30:
yes, I know.
I assume trainer would work if the headtracker behaves like the old module.
Whatever that means, I am no bluetooth expert.

rotorman, 2021-04-30:
Found a long docu about OpenTX for FrSky and clones and it lists on page 836 CC2650 as PARA chip:
[https://www.schorse-video.de/images/webshare/OpenTX_Manual_Deutsch.pdf](https://www.schorse-video.de/images/webshare/OpenTX_Manual_Deutsch.pdf)

dlktdr, 2021-05-20:
Well Guess what...

# I got a connection! 
**"Trainer signal recovered" never sounded so good.**

I almost was about to give up and didn't realize it was actually receiving data!

I've been porting all the code over to Zephyr OS. It's Bluetooth library is more mature and also supports pairing out of the box.

I know probably everyone with this board has probably already swapped it.. but there is hope for those that haven't!

Because I have to ditch all the Arduino code and have to re-work parts of it it's going to be a bit before a working copy is ready. So far just serial and Bluetooth are working. Initial tests are it's a far better operating system. Just the support for thread pre-emption is worth learning it. 

I have high hopes with this setup there will be no disconnects and I can up the update rate now that it can interrupt other parts of the code easily.

rotorman, 2021-05-20:
I am still stuck with the "cheap" CC2540, so this is absolutely awesome news! For my understanding - the encrypted connection with CC2540 works under Zephyr, but didn't under Arduino? (pairing code likely 000000 ?) Best!

dlktdr, 2021-05-20:
So there is no passcode after all.

In Zephyr you can set the characteristic permissions to be BT_GATT_PERM_WRITE_ENCRYPT or BT_GATT_PERM_WRITE_AUTHEN. The latter which requires an actual authentication. Turns out it's just encrypted so sniffers can't see the data hence why we can see it on our phones without a code. Learning more about BLE as I go...

I have to admit, I didn't try too hard on the modified Arduino library with pairing support so it might have still been able to do it. I really wanted to solve the other Bluetooth issues so I thought I would give Zephyr a go.. so far so good. And since I still have the cheap CC2540 module in both of my radios right now this was my first try.

Currently working the unforeseen issues. Now that threads can be pre-empted has caused a bunch of non-thread safe & re-entrant code to fail. Also have to re-work the I2C libs, Flash and remote para still. Luckily since I did PPM, SBUS and PWM on hardware level they work as expected :)

rotorman, 2021-05-26:
It looks FrSky is not making, at least CC2540 boards, themselves but get them from Shenzhen RF-star Technology Co., Ltd.:
https://www.szrfstar.com/upload/file/1586418083.pdf (page 2).

CorruptName, 2021-06-08:
So I found this thread because I purchased:
[https://www.horusrc.com/en/frsky-horus-x10-para-wireless-module.html](url)
After watching AndyRC's video.
I have bluetooth address but the name wont change from "hello" and I'm not able to connect to it.
Any Suggestions?

dlktdr, 2021-06-08:
So with the para module the name is always hello. That's normal no matter what you set it to on your radio.

Just to check you have your radio set to Bluetooth trainer under the hardware page. Under the model page you should choose **Master / Bluetooth** at the bottom.

GUI needs to have the **Bluetooth mode enabled** on the bluetooth tab first.

Then choose **Discover** option below Master/Bluetooth on your radio, the Address of the Headtracker should be pop up in a few seconds and match the address the GUI shows.

CorruptName, 2021-06-08:
Thanks for the help. Do you know if telemetry works over this module as
well? Or should I get a better module? If so, which one?

dlktdr, 2021-06-08:
Afaik it does, haven't personally tried tho. Pretty sure the PARA one is their most recent module. Did you manage to get it to connect?

[7 more replies]

closed 2021-06-13 by dlktdr