headtracker/HeadTracker #23: Sbus support for para master

SilentZen, 2021-03-21:
Would it be possible to add sbus support to para master, this would allow module to be plugged into s-port on x9d and use the sbus trainer option.

dlktdr, 2021-03-21:
I was actually thinking about sbus anyway. Yes it's possible. Could have a 3d printed jr bay you can just plug in.

dlktdr, 2021-03-21:
Here you go.. SBUS output on TX pin on Para Master.
[image]

It needs testing still but. you can you give it a go. Unfortunately I can't invert it in software easily. Have an idea tho using the PPI that might work.

[ParaMaster v1.3.bin.zip](https://github.com/dlktdr/HeadTracker/files/6178443/ParaMaster.v1.3.bin.zip)

dlktdr, 2021-03-22:
Inverted output available now. Using nordic's PPI to invert it worked

Still requires testing.

[ParaMaster v1.3 wSBUS.bin.zip](https://github.com/dlktdr/HeadTracker/files/6178704/ParaMaster.v1.3.wSBUS.bin.zip)

[image]

SilentZen, 2021-03-23:
Thanks, that was really fast.

dlktdr, 2021-03-24:
Well luckily someone had already wrote the SBUS library. Took me longer to find out that the arduino on this BLE has a bug setting the UART to 2 stop bits and 10000baud causing a crash + figuring how to invert it.

For some reason I can't get it to read on my taranis. Even though it looks identical to a capture of SBUS out on the scope, even decodes correctly on the scope, including timing and levels, parity.. to the SBUS coming out of my remote.

If you get a change give it a go, I've never used sbus trainer before figure it would probably just say "trainer recovered" like the other methods.. leading me to believe there is something not quite right still or I have my radio setup wrong. If it works for you awesome if not I'll pull a receiver from a plane with sbus out to test with when I get a chance

Riffer, 2021-03-27:
To monitor the signal one can just set up an old flight controller with INav or Betaflight. I used it in the past to check Arduino generated PPM.

Riffer, 2021-03-28:
@dlktdr No successful testing. 

After trying out my Radiomaster TX12, TX16S and the FrSky X9D plus I did what I mentioned above.
Set up a monitoring with a flightcontroller (F3 SRF3Pro) and to be absolutly sure, I first connected a standard SBus-Receiver to the exact same pin and checked in iNav that the configuration works - and it did. But either RX nor TX of the Arduino Nano 33 BLE worked. 

Something must be wrong. Did you ever try it successfully?

dlktdr, 2021-03-28:
@Riffer  No I haven't had a chance to get back to it yet. Wasn't working for myself either.. 

I've got the two remote and head firmware combined now and added SBUSout support to the main FW. So now all features will be available on both. Makes supporting a single file easier now too.

Have a few more things to go on this, then I'll go find myself a receiver/flight controller I can test with.

I must have the baud rate off a little bit or could me misusing the library. I can try the SBUS lib from ardupilot too, the code actually looks a bit cleaner anyway.

Riffer, 2021-03-29:
@dlktdr If you have anything to test, just let me know. One could use another device (a nano clone or something) to convert from PPM to SBus but I feel that would be overkilling.

Maybe that gist helps: https://gist.github.com/reefwing/265bafc35b47b7c813743add89dc0688 ?

dlktdr, 2021-03-29:
@Riffer That would have helped a lot.. I calculated from the NRF data sheet,

```
#define BAUD100000                   0x0198EF80
#define CONF8E2                      0x0000001E
```

but I see they found, 

```
const uint32_t CONFIG       = 0x000000F;
const uint32_t sbusBaudRate = 0x19114A7;
```

Baud rate is slightly different and the conf register is too.. Little confused about the conf r

dlktdr, 2021-03-31:
Semi Success!!

Well turns out the settings I had figured out work out more accurately to 100000 baud & the stop bits were wrong with config set to 0x0000000F were correct at 0x1E.

Still not sure why I can't connect it to my Taranis but I was able to make a connection to my Mateksys 765 Wing & iNav. Has to be on the SBUS Normal (RX) Pin.

[image]

As for Taranis starting to wonder if it's something in my radio. I took a sample SBUS output from the radio, got the code to match exactly (took mirror images of both on the scope and they were a dead match).  Passed it back in and no go, won't read it???

Give this one a try if you can. See if your flight controller can read it, and your radio?

[PARA1.3_SBUS_TEST.bin.zip](https://github.com/dlktdr/HeadTracker/files/6234075/PARA1.3_SBUS_TEST.bin.zip)

Riffer, 2021-03-31:
Good news first: It worked at my X9D plus, see picture - this is with the PARA-Module connected to the module bay - I mapped TR1-TR8 to channels 10 upwards.

[image]

Bad news - somehow I killed the module power regulator (USB powering still works) on the Arduino while I tried to get it working with the Radiomaster Serial RX Port. I was somehow suprised that the JR module bay no longer works for SBus input - the menu does not provide it, but if one sets the TTL Uarts in the hardware menu there might be a way to get it working on the external UART - but future will tell.

Riffer, 2021-04-01:
From your description I guess we did it the same way, but here are two pictures from mine.

The 2019 has another CPU - and I see this Issue on OpenTX: https://github.com/opentx/opentx/issues/7849 - maybe you just add your two cents?

[image]
[image]

Riffer, 2021-04-02:
@dlktdr maybe you check the CPPM input instead? Just an idea - I am not sure 2019 version has this option for the module bay in OpenTX, my 2015 version has it, but I did not test it. I fail using the Radiomaster TX12 with SBus-Master in the modules bay - maybe the same bug as yours.

SilentZen, 2021-04-05:
Details for Taranis X9D
https://github.com/opentx/opentx/wiki/Taranis-IO-ports
Need to use pins 1 (Ground) ,2 (VMAIN)  and 4 (UART_R-SBUS_INPUT)

How to setup on radio
[image]

Picture of  receiver using port

[image]

dlktdr, 2021-04-09:
Posted Firmware to Master

closed 2021-04-09 by dlktdr

SilentZen, 2021-05-08:
> I have a question hopefully somebody can answer ,I’m putting in a BL 33
> board for head tracker on a Taranis X9D but I also want to enable wireless
> trainer connection should I use pin three to transmit trainer instructions
> from that radio. Dan ***@***.***

Actually ignore pin 4 aandcrx us for sbus trainer.

kiwi41000, 2021-05-08:
OK now I’m really confused one response says use pin 4 for the other response says ignore it. I just want my head tracker to work right now. Then I will work on wireless trainer. Dan

SilentZen, 2021-05-08:
If you want to use sbus via wire it's pin 4 on x9d and Rx port on nano33ble
(t16 uses tx pin). You mentioned wireless which threw me.

kiwi41000, 2021-05-08:
OK I’m looking to get that wireless head tracker working right now and then later wireless trainer capability. Dan

dlktdr, 2021-05-08:
Hi Dan,

Just trying to understand here. Your trying to get the wireless trainer from a student as well as a wireless head tracker at the same time?

kiwi41000, 2021-05-08:
Not at the same time wire less trainer system later. I’ve got the 2019 Taranis and will be receiving from that should work for both. Dan

dlktdr, 2021-05-08:
Ok, If you have the Taranis X9D Plus 2019, (**Not SE**)

If you want to use SBUS on Radio you should hook either TX or RX pin from remote board into the pins shown above. One is inverted one isn't. Just try both until you hear "Trainer Signal Recovered" on the radio. You should still hear this even if blue tooth isn't connected to the headtracker

(BLE) - > (Radio, SBUS in battery bay as shown in above photo)
```
Gnd->Gnd (pin 1)
TX or RX->UART_R-SBUS_INPUT (pin 4)
Vin-> VMain (pin 2)
```

**Note** I was never able to get my SBUS trainer to work on my X9D Plus 2019 SE version, it's possible if your not receiving any channels it's the same issue as the non SE version. In which case use a PPM cable like in video here https://www.youtube.com/watch?v=QDammUCZmpo

kiwi41000, 2021-05-12:
I’ve run into a new problem that I can’t figure out I have 2BLE 33 boards going to be used for head tracking on a Taranis x9d subs mode. Both of them Upload and connect properly when I use version .9.91.92. When I try to change them to the para master version 1.2 1.3 or 1.4 they upload firmware properly but they will not connect with the caption unable to determine board type what am I missing been working on this all day. Dan

dlktdr, 2021-05-12:
So one board should be programmed with .9xx The second one should be programmed with the para master, which is the one that's going to output to your radio BUT

You can't connect to the PARA master firmware from the GUI. It is a very basic program that just connects to the first available head unit and starts outputting PPM or SBus signal pin numbers are fixed. The light on it should go blue when it connects is the indicator it's connected

I put the note right after item 10.. I guess I should make it bigger You're not the first person to ask.
https://github.com/dlktdr/HeadTracker/wiki/Nano33BLE-Build-Instructions#programming

kiwi41000, 2021-05-12:
So that means the Bluetooth is enabled already on the parra unit. And if that’s the case what do I do to Bluetooth them together. Thanks for your response I thought they were broken or I was doingThanks for your response I thought they were broken or I was doing something wrong.  Dan

kiwi41000, 2021-05-12:
OK I have blue lights on both units I assume that this is working. Dan

kiwi41000, 2021-05-12:
Yes you are correct there is a note after item 10 my bad completely thank you for your patience. Dan

dlktdr, 2021-05-12:
All good.. I do need to make that clearer, but since it's changing in 1.0 I was going to wait a do a more complete re-write.

Message me on Discord chat group if you need more assistance in the setup.

kiwi41000, 2021-05-12:
Thank you they are both working great

kiwi41000, 2021-05-25:
The head tracker works great with my trans 2019 but I’m still having problems with the settings in the radio for the Taranis ACCST .in the mixes menu what should be the source for that channel. Dan