headtracker/HeadTracker #24: Add PWM output A0-A3

dlktdr, 2021-03-27:
I'm thinking of adding 4 channels of PWM output on a0-a4. User can select which variable they want on the pin, choices are Bluetooth inputs 1-8, PPM inputs 1-16, SBUS nput 1-16, Tilt, Roll, Pan. Any others?

PWM also takes up very little processing power so won't affect the main purpose of this board.

Is this a bad idea, comments?

Riffer, 2021-04-06:
If this is on the receiver part, then it could be used for near-range remoting of a cam or a bot. 
But which scenario do you have in mind? The sending part as well as the receiving part is nearby the pilot and not at the model?

dlktdr, 2021-04-07:
Goal of the next firmware release is to have both receiver and transmitter on same FW so all features will be up to you what board they are on.

The user who had the head tracking spotlight idea could control the motor directly. Have another idea I want to try, active balancing an object. Will take a video if it works :) Can think of a few but not too many. Maybe a short range remote controller (flight stick idea).  Active horizon gauge with a servo. So no not too much related to head tracking.

I know I could just get a PPM to PWM converter but why do that when the board is more than capable. I would do this all low level in the NRF PWM peripheral. Only way I see to get the 1us resolution, fast 20ms updates & very minimal CPU. I'm not too worried about this or the analog idea getting in the way of head tracking related, most people will just leave disabled so will only cost 4 if statements, but if you have an idea go ahead and turn it on.

closed 2021-04-09 by dlktdr