headtracker/HeadTracker #26: Unable to determine board type moving from GUI .72 to .90

kevin-kevjan, 2021-03-29:
BLE has firmware 0.72 loaded...tried to connect using .90 GUI to update and got "unable to determine board type" error.
Connecting with GUI .72 connects successfully.
Screenshots below.

Serial info screen shows:
`Connected to COM6
Trying to connect to NANO33BLE 
GUI: {"Cmd":"FW"}� 
HT: FW Requested 
Trying to connect to BNO055 
GUI: $HARD 
GUI: $VERS 
Disconnecting from COM6
`

[image]
[image]

kevin-kevjan, 2021-03-29:
**Upgrading firmware on board to Ver 0.90 does resolve issue.**

dlktdr, 2021-03-30:
Since 0.8 old firmware will not connect due to the addition of CRC checking of data. 

Always try to be sure you are on a 0.9x firmware and a 0.9x GUI, or a 0.8x Gui and 0.8x FW to be sure of compatibility

closed 2021-03-30 by kevin-kevjan