headtracker/HeadTracker #27: Add SBUS output option

rotorman, 2021-04-03:
Enhancement idea: with RadioMaster TX16S, one could power the Arduino BLE33 and pass the signal directly via one of the bottom serial AUX ports, when the output encoding would be in SBUS instead of PPM (In OpenTX Serial port setting: _SBUS Trainer_.

[image]

dlktdr, 2021-04-03:
Duplicate of #23 

Already complete on remote board. In testing phase. See link for firmware to try on one of my later posts

closed 2021-04-03 by dlktdr

Riffer, 2021-04-04:
I already tried this setup but failed to get any signal into the TX trainer master - (I thought I fried my Arduino at that very moment, but after a day, it works again - maybe there is a resettable fuse in it, who knows - and the AUX are directly wired to the CPU. Not a great feeling when lights go out... (not the TX but the Arduino). 

These ports of the RadionMaster TX16S are not so well documented in regards to OpenTX as I wish.

To stress out: The settings in OpenTX tell more than any documentation anywhere else at all. OpenTX is a nightmare in this matter - you always have to look at different places, dig up issues from github and end up studying source code to get a clue how a feature should work. Then, to find out that it does not work as expected or has limitations - or let them call bugs - one example may be that the two different ports are used from LUA as the same - it seems you need to guess which is the right one.

As I mentioned, my setup with RadioMaster TX16S did not work. @rotorman, will you test the "SBUS Trainer" that @dlktdr  mentioned? You find the current PARA Master there; just flash as a local version from the GUI. 

I succeeded in making the same version work at my 2015 X9D plus in the module bay. And I will give the RadioMaster another try with just an RX in SBus mode to check out that it works before killing an Arduino or the TX.

rotorman, 2021-04-04:
Just ordered one [Nano BLE 33 Sense](https://store.arduino.cc/arduino-nano-33-ble-sense) board for testing, thus hope to be able to test this also in a wired SBUS output mode (as in Option 2, but with SBUS protocol and connected via DuPont 4-pin socket to TX16S AUX1|2 instead of 3.5mm jack):

[image]

SBUS has IMHO the added benefit of being more robust as being true digital communication signal (UART 100000 baud 8N2), when compared to PPM that has binary levels, but otherwise is an analog signal - pulse length is being timed/measured.

Riffer, 2021-04-04:
I need to add this one: I fried the AUX on my Radiomaster. That just happened in the above-described situation; I wanted to test the same situation again with an SBus RX and whenever I try to switch to the Option the TX goes into "EMERGENCY MODE".
Ok, lessons learned - I will see whether I can find a European retailer for the mainboard and will also solder the 30$ module on it (already ordered one just in case).

rotorman, 2021-04-04:
The RX/TX pins of TX16S are connected to main STM32F429BI µC via 100R resistors. As long as the Nano did not put considerably more than 3.3V or less than 0V onto the pins, it should not have an effect on destroying the mainboard. I have done some work towards the schematic of TX16S, here first the pinout mapping of the main µC:

[image]

You can find more about my TX16S shenanigans on my [RCGroups blog](https://www.rcgroups.com/forums/member.php?u=863799).

I guess the [preliminary SBUS firmware](https://github.com/dlktdr/HeadTracker/files/6234075/PARA1.3_SBUS_TEST.bin.zip) by Cliff (@dlktdr) in #23 is only for Option3, right? So I cannot test this with only one Nano 33BLE?

[image]

dlktdr, 2021-04-04:
@rotorman You can give this one a go, [BLE v1.0 _ SBUS .bin.zip](https://github.com/dlktdr/HeadTracker/files/6254588/BLE.v1.0._.SBUS.bin.zip).  It's a current compile from the combine FW branch.

This is going to be the firmware with BLE Head+Remote, SBUS, ANALOG and PWM.. so there is lots un-tested and broken, but the basics, including SBUS work. I just tried with 0.9GUI and the SBUS signal works to my Matek765. Same pins as Remote board (tx + rx)

rotorman, 2021-04-07:
I can report back good news. The BLE v1.0_SBUS in your last post works beautifully, together with RadioMaster TX16S.

I flashed it to my Arduino Nano 33 BLE Sense. Enabled for channels 9,10 and 11 and connected the Nano 33 BLE Sense using three wires to TX16S AUX1 - GND to GND, VIN to +5V and TX of Nano (not inverted SBUS output) to TX of TX16S. TX16S AUX1 RX unconnected.

I set in OpenTX SYSTEM -> HARDWARE -> Serial port (TTL) to SBUS Trainer.
In MODEL -> SPECIAL FUNCTIONS, configured SF1 to SC up as Trainer Chans and enabled it.
In MODEL -> MODEL SETUP -> Trainer, set Mode to Master/Serial.
By going into (long roller press) Monitors -> Channels Monitor 9 to 16 and flipping Switch C up, I can nicely see channel 9, 10 and 11 match the head tracker = Arduino Nano 33 BLE Sense orientation.

Awesome, thank you very much Cliff for this wonderful project (also the original authors Dennis and others!)

dlktdr, 2021-04-07:
I did notice on that FW, I had the SBUS output un-invert on me while it was sitting on my desk. It was running for a few hours, not sure when or why it happened. It's using the PPI interface to toggle the normal based on the not-inverted output, so if there is a real slow down, maybe a noise on the pin it will flip, or a software issue... Have an idea to fix to be sure it gets set properly on every sbus frame, haven't got there yet.  I had Bluetooth on too, which is a processor hog might have caused it.

Just be sure to have an override switch if your intending to fly with it.

rotorman, 2021-04-07:
Good to know that there might be a glitch to watch out. I had BT off for my (brief) testings.

dlktdr, 2021-04-07:
Yeah I guess that's true, most things can read normal UART levels these days. Might be a good idea for me to suggest to everyone use the not inverted TX pin if capability available. If its a continued problem I'll just disable.

Will do some more testing. But I think it should be just, disable ppi, set inverted pin, re-enable PPI before uart start. Then no chance or just a bad frame if something goes wrong.

I would really like change this to use just go low level UARTE peripheral with easyDMA. It's a blocking serial call through Arduino + MBED interface and bloated for sending only 24bytes every frame. This would be an ideal use and no cpu-invervention except quick interrupt for updating the values.

There is a little jitter while the sbus frame data comes out, would solve that too. Not that it matters too much for UART tho.

rotorman, 2021-04-07:
Maybe this helps to take a bit of load off from you (if you find it OK, feel free to use/modify it as you like. I can also send you the CorelDRAW 2018 original if you want):

[image]

dlktdr, 2021-04-07:
OOoo nicer than I would have done. I Like it. Don't think I need to modify it.

But if you have send the file, I'll put it in the docs folder. I also used Corel for the others. Just pushed them to the Master Docs folder too..

rotorman, 2021-04-07:
Is there a way to send me your email? Maybe via PN of RCGroups?