headtracker/HeadTracker #30: Idea for improving flash handling

Riffer, 2021-04-10:
Currently we need to save before and restore configuration after flash, but maybe you can change it in a way that the GUI temporarily keeps the configuration automatically for flashing and restores it afterwards.

Another idea is that you start update with a soft reset into boot mode before flash (as platformio does, I guess the code should be available), instead of having the users make a reset by button press. This way we can put the Arduino in a fixed case with only a hole for USB.

dlktdr, 2021-04-10:
> Currently we need to save before and restore configuration after flash, but maybe you can change it in a way that the GUI temporarily keeps the configuration automatically for flashing and restores it afterwards.

Yes that's doable.. would make it easier to update. I've gotton so used to just loading my config file everytime, would save a step. Not

> Another idea is that you start update with a soft reset into boot mode before flash (as platformio does, I guess the code should be available), instead of having the users make a reset by button press. This way we can put the Arduino in a fixed case with only a hole for USB.

I tried this using just bossac and the -a argument and drop the -port argument when calling.. but if you have more than two ports it won't work and if you specify the comport it doesn't go back and look for the new one. 

Looks like it can be done I see the py script is in, .platformio\platforms\nordicnrf52\builder\main.py where it forces 1200baud. then looks for a new port.. then calls bossac on the new port. Not sure where `TouchSerialPort` is located to see how they are doing it..

I would probably do a wizard, 
- Pick your firmware on first window then click next (With check box choice to restore settings)
- Show a message saying plug in the board(or disconnect and re-reconnect 
- Auto-detects it, 
- use 1200b hack to force bl.
- wait for new port
- write with bossac
- upload settings (if checked)
- show complete or failed message

Sounds like a lot of work but I can see how it would be helpful.. helped someone on the chat this morning who had that issue, forgetting to leave a hole for the reset button.

dlktdr, 2021-04-29:
This is now complete. Will be in v1.0 Minus the auto uploading settings which I'm not going to worry about for now. You can now flash without hitting the button.

[image]

[image]

dlktdr, 2021-05-10:
Posted a test copy here. 

https://github.com/dlktdr/HeadTracker/wiki/Version-1.0-RC1-Beta-Test

closed 2021-05-18 by dlktdr

Riffer, 2021-05-23:
@dlktdr 
Hello Cliff, it took some  time to test:

* Flash works very well, even with a brand new device

But while calibration the USB connection breaks every sigle time. I suspected a problem with my cables first, but after about 4 tries I went back to 092 and the one works splenditly. Maybe a coincidence but now I am only happy that my device works with the older version again.

dlktdr, 2021-05-23:
Did the board stay running, after it disconnected (lights still blinking)? Wondering if it crashed e.g. Firmware issue or a GUI issue.

Were you able to reconnect without rebooting the NANO?