headtracker/HeadTracker #32: Board orientation

a1axx, 2021-05-09:
Using the Fatshark goggle module door setup, 2 axes are mixed as orientation can not be customised.

dlktdr, 2021-05-09:
Have this fixed in 1.0 so you can manually enter the orientation to solve this issue.

If I can just get the Bluetooth to stay connected I'm going to release it..

dlktdr, 2021-05-10:
Are you planning on using Bluetooth wireless functionality? If not, I can send you a current copy you can try. Bluetooth is not stable on it yet.

dlktdr, 2021-05-10:
Fyi you are probably going to want to set,

Tilt 90 OR -90 and then Pan 30 Or -30 , leaving roll(y) at 0. For the way you have it setup. ...Assuming it's 30 degrees off from center.

When it does the rotation it does x (tilt) first, then y(roll) around the new orientation, then z(pan) around the next new one

dlktdr, 2021-05-20:
@a1axx Have you had any luck with V1.0?

a1axx, 2021-05-22:
Giving it a wee try tonight

a1axx, 2021-05-22:
> @a1axx Have you had any luck with V1.0?

Hi mate, Works perfect!

-90 Tilt and -30 Pan worked fine

closed 2021-05-22 by a1axx