headtracker/HeadTracker #34: RC1.0 Drift

a1axx, 2021-05-25:
Hi 

https://www.youtube.com/watch?v=g1SLBUuEn3I

Video above to compare RC1.0 -> 0.92

The drift is subtle but consistent, after calibrations and resets the drift persists, the axis affected here were Pan & Tilt. Roll appears to be okay.

When flying, I was flying LR on Sunday I was actually turning in circles to counter the constant drift on Pan. I ended up just switching over to sliders. 

0.92 is totally rock-solid, I sometimes have to turn my head to check it's still working as it doesn't drift!

All settings default

dlktdr, 2021-05-25:
Thanks for the video. I see what you mean.

I'll take a look at what could have changed in the IMU code between the two.. I didn't intentionally change anything there. I'll take a more detailed look when I'm home from work tonight.

a1axx, 2021-05-25:
> Thanks for the video. I see what you mean.
> 
> I'll take a look at what could have changed in the IMU code between the two.. I didn't intentionally change anything there. I'll take a more detailed look when I'm home from work tonight.

Yea no worries at all, at the mo it works ace on 0.92, I am sure there will be a lightbulb moment when you see the issue!

Classic, fix one thing, break something else lol

dlktdr, 2021-06-13:
Think I found the issue, was rounding the values to a lower precision. Fixed in 2.0

Will re-open if required

closed 2021-06-13 by dlktdr