headtracker/HeadTracker #36: jeti users

jag3957, 2021-06-11:
possible connections modes for jeti users. PPM TX physical connection NOT must be more that 3.2 to 4.2 max 1A. arduino is 5 to 20 volts.

dlktdr, 2021-06-11:
Not sure I understand what you asking

Are you using Arduino nano 33 BLE? It is only 3.3v on the PPM output pin.

Input voltage to the board is 5-20 which is unrelated to the IO pins. There is a regulator on the board that drops that voltage to 3.3 for everything

jag3957, 2021-06-11:
so I can connect directly to pins 1,2,3 in the tx?
[image]

dlktdr, 2021-06-11:
As long as you using the Arduino nano 33 BLE. Not the BN0055 project on this site too. Then yes you can direct connect them they are 3.3v in and out.

jag3957, 2021-06-11:
thank you too much. yes I found in amazon. I will try to make it wireless that is my dream LOL. jeti have a wireless trainer mode , they connect a second TX to a primary TX and sent finally to receiver using just the primary TX. I am dreaming to connect the nano 33 to a jeti receiver and configure the pin in the receiver to receive and sent those commands to primary TX finally sent it to aircraft receiver in the plane. you know if can do that?

dlktdr, 2021-06-12:
Not too sure. Will have to try.

closed 2021-06-12 by dlktdr