headtracker/HeadTracker #37: Analog Read Reference Voltage

dlktdr, 2021-06-13:
Get it set to VDD, currently using internal reference, causing voltages readings to max out at 0.6v

rotorman, 2021-06-13:
Possibly the following PDF of Nordic can help: Sec. 6.23.2 (page 379) and 6.23.9.20 (page 394):
https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.2.pdf
The solution could be to set the ADC reference to VDD/4 and gain of 1/4.

dlktdr, 2021-06-13:
Think I tried that but I'll try again. 
I'll have to dig deeper into the Zephyr Analog Nordic Driver., if set to the internal it works as expected, set to any other source no reads..

Just making a list of known issues for now, working on getting the zephyr version release ready to go. So people know what doesn't work.

dlktdr, 2021-06-14:
Completed.

dlktdr, 2021-06-15:
Posted in 2.01

closed 2021-06-15 by dlktdr