headtracker/HeadTracker #41: Receiving data via CC2540 constantly disconnects - Remote Board

dlktdr, 2021-06-13:
Currently when using a FRSky BLE module as the Transmitter and the Nano33BLE as a remote board the connection is highly unstable. Constantly disconnects..

This as important of an issue, unless your trying to use a Nano33BLE as a trainer bridge to another non-frsky radio?

robthomson, 2021-08-24:
I have observed the same issue with the use on my jeti radio.

System disconnects and reconnects almost instantaneously - but result is momentary jerkiness on the pan servo.

For now I have gone back to using the cable.

dlktdr, 2021-08-24:
@robthomson I don't think this issue is related to what your mentioning. This issue is when using the Nano as a receiver from a old Frsky BT module. This CC2540 chip is this one found here. https://www.horusrc.com/en/frsky-transmitter-horus-x12s-bluetooth-module.html.

I assume you are using Option 3 for hookup. Might want to check out #53 and #48 first to make sure there isn't a BT range issue. Can open up a new issue with what your seeing if your still having issues.

dlktdr, 2022-01-26:
I'm going to close this as I don't intend to fix it. Not the primary purpose of this project to receive data from a radio.

closed 2022-01-26 by dlktdr