headtracker/HeadTracker #44: Possible Issue with desktop application

CorruptName, 2021-06-14:
I updated to the new version 2.0 last night and it seems to be working great.
When i went to update the firmware through the application I selected "Online V2.0" and there is no firmware available. I had to select "Shipped Firmware" before the listbox would fill. Just thought you should know. Once you select Shipped, you can then select Online and it will work (it seems)

[image]

Thanks again for your awesome work.

dlktdr, 2021-06-14:
Good to know. I'll see if I can reproduce it

dlktdr, 2021-08-22:
Haven't been able to reproduce here. Maybe a network connection issue startup?

Going to close this out. Let me know if I need to re-open

closed 2021-08-22 by dlktdr