headtracker/HeadTracker #45: CC2540 Non-PARA Bluetooth module doesn't reconnect after radio reboot

dlktdr, 2021-06-17:
I'm not sure exactly what's going on here yet but if you restart the radio it won't try to reconnect to the Nano. If you go to BT and choose discover again it will reconnect.

It doesn't seem to be trying to make a connection attempt, not sure if OpenTX isn't saving the address or there is some kind of issue with it wanting an advertising response maybe? 

If the headboard reboots while transmitter is on everything reconnects normally.

Just click discover again for now. At least it's sending data :) I will look into it.

ethomas997, 2021-06-17:
The issue I'm seeing seems like kind of the opposite.  If the headtracker is on when I power on the QX7S then the connection is made successfully.  If the headtracker is off when I power on the QX7S, and then I power on the headtracker, the connection is not established.  At that point, cycling power on the QX7S will work to make the connection, or doing "discover" in the QX7S menu will work.

Using Nano 33 BLE (not "sense" version) and Taranis QX7S ACCST (non-PARA) OpenTX 2.3.9.

dlktdr, 2021-06-25:
Think I've solved this one.

It was advertising with the random address, so if the random address changed the radio couldn't find it again. And explains also why it was different issue for everyone.

Added this to the KConfig (prj.conf) options
`CONFIG_BT_SCAN_WITH_IDENTITY=y`

Have to verify it didn't break PARA still.

dlktdr, 2021-06-25:
Okay now I'm confused. Connecting very consistently on the old versions too...

ethomas997, 2021-06-27:
If you have a build I can try I'll test to see if it clears up the issue I was seeing.

dlktdr, 2021-06-27:
This is a current compile of 2.10. I started modifications to support of SBUS input and added an extra analog channel and some minor changes. so not all features in the GUI in 2.0 you have will work. It will also complain that firmware is too new, just ignore it.

I haven't been seeing the same connecting issues on this one like I was before. I can get it to crash if I power the radio off, don't wait long enough for the blue light to go out and power it back up. Other than that seems to reconnect no matter which one is reboot.

[firmware.zip](https://github.com/dlktdr/HeadTracker/files/6721151/firmware.zip)

ethomas997, 2021-06-28:
I tried that v2.10 firmware but saw no change; still behaves as described in my first comment above.  I upgraded the OpenTX on the transmitter to the latest version (2.3.13), but no change there either.

I guess it could be an issue on the transmitter side. Is there a way we could try to determine that?

SilentZen, 2021-06-28:
I have para bluetooth module and I have noticed the trainer setting in
model profile does not always seem to store the bluetooth address of the
headtracker after a reboot. May be worth testing reboots with trainer in
range vs reboots with trainer out of range and see if it removes it if not
found.

ethomas997, 2021-06-28:
From what I'm seeing, if the headtracker board is powered on first then the QX7S transmitter always seems to successfully connect to it.

dlktdr, 2021-06-28:
@ethomas997 I wonder if you could see if it's still advertising when it won't connect. If you get it into the state where both are on but not connecting can you use your phone and a Bluetooth scanner (NRFConnect or BLE scanner android) app and see if you see the "Hello" device. If you have that you could try connecting from your phone and see if the blue light comes on.

If it's advertising and able to connect, probably the radio. If not something on my end.

ethomas997, 2021-07-03:
It looks like the headtracker is advertising itself when it comes on.  It seems like the QX7S ACCST bluetooth will look for the headtracker once when the QX7S powers up, but then not again after that.  

Would you expect the master to keep doing periodic scanning for the headtracker?  Is that what you see happening with the PARA radios?

dlktdr, 2021-07-03:
Yeah the PARA if it's not connected will always scan and try to connect again.

My CC2540 seems to do the same now and I'm not sure what I did. I was having a heck of a time getting it to reconnect after radio reboot then all of the sudden re-connects like a charm, no matter what's shut off. Although not what your seeing of course.

@rotorman  sent me a BLE scan of the CC2540 doing this, I'll look it over in more detail to see if there is some more hints. It looks like the Nano is advertising fine, but the CC2540 just stops scanning OR is scanning and just sees something it doesn't like and won't connect.

dlktdr, 2022-01-26:
I haven't tried this in a while is anyone still having issues with CC2540 module. Can I close this out?

ethomas997, 2022-01-26:
As far as I know the problem I described is still present.  (Haven't used the module since the cold weather hit.)

dlktdr, 2022-05-30:
Discovered this is an issue with the way OpenTX and the CC2540 module work. The Para module trys to reconnect to last as soon as it's powered. The CC2540 doesn't and Edge/Open doesn't request a re-connection so it never tries again until you select it.

This will need to be solved on the radio. See https://github.com/EdgeTX/edgetx/issues/1204

dlktdr, 2022-12-28:
Issue is still present,

[OnRebootNotWorking.txt](https://github.com/dlktdr/HeadTracker/files/10315625/OnRebootNotWorking.txt)
[Working.txt](https://github.com/dlktdr/HeadTracker/files/10315626/Working.txt)