headtracker/HeadTracker #46: Functions.cpp:112:15: error: 'FOC0A' was not declared in this scope

waldher, 2021-06-26:
I'm finding a number of constants that are supposed to be defined, but are not. I can't find these in an include anywhere. Is there a bug, or am I missing a configuration option when building?

```
Functions.cpp:112:15: error: 'FOC0A' was not declared in this scope
(0 << FOC0A)| //
^~~~~
Functions.cpp:113:15: error: 'FOC0B' was not declared in this scope
(0 << FOC0B)| //
^~~~~
```

dlktdr, 2021-06-26:
This is for the older unsupported BNO055 code. You probably have the board set incorrectly. The older one should be set as Arduino Nano. https://github.com/dlktdr/HeadTracker/wiki/BNO055-Build-Instructions-(Discontinued)

Is that what your intending to compile? The newer Nano33BLE code should be compiled using VScode and platformIO and is under https://github.com/dlktdr/HeadTracker/tree/master/firmware/src/Nano33BLE_Zephyr 

I will quickly write a guide if you want help compiling this there are a few changes that are not obvious when starting out with zephyr and platform io.

dlktdr, 2021-06-26:
Please See For a Compile Guide on the New Zephyr Version

https://headtracker.gitbook.io/head-tracker/support/development

closed 2021-06-26 by dlktdr

waldher, 2021-06-26:
That's perfect. This resolved the issue, thank you!