headtracker/HeadTracker #47: The custom arduinoBLE dependency is missing

omegat, 2021-06-29:
Hi,

I tried to build the code, but PIO is complaining about not finding https://github.com/dlktdr/HeadTracker.git 
And... it's not there. Is there any alternative to use?

Cheers, Tobi

rotorman, 2021-06-29:
Hi Tobi,
you either need to git clone the repo locally, e.g.
git clone --recursive https://github.com/dlktdr/HeadTracker.git
or download the zip https://github.com/dlktdr/HeadTracker/archive/refs/heads/master.zip and unpack it. If you still do not proceed, plz post a screenshot of the PlatformIO showing error.

omegat, 2021-06-29:
Hi thanks for the quik answer!
I cloned the repository and tried to build it using the commandline:

`PS D:\...\HeadTracker\firmware\src\Nano33BLE> pio run
Processing HeadTracker_Nano33BLE (platform: nordicnrf52; board: nano33ble; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/framework-arduino-mbed @ 1.3.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduino-mbed @ 1.3.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nano33ble.html
PLATFORM: Nordic nRF52 (8.1.0) > Arduino Nano 33 BLE
HARDWARE: NRF52840 64MHz, 256KB RAM, 960KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:
 - framework-arduino-mbed 1.3.2
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Library Manager: Installing arduino-libraries/Arduino_APDS9960 @ ^1.0.3
Library Manager: Arduino_APDS9960 @ 1.0.3 has been installed!
Library Manager: Installing git+https://github.com/dlktdr/ArduinoBLE.git
git version 2.16.2.windows.1
Cloning into 'C:\Users\tobia\.platformio\.cache\tmp\pkg-installing-toayu6uw'...
remote: Repository not found.
fatal: repository 'https://github.com/dlktdr/ArduinoBLE.git/' not found
[...]`

I then cloned it again /w --recursive as you suggested, but:
`PS D:\...\HeadTracker\firmware\src\Nano33BLE> pio run
Processing HeadTracker_Nano33BLE (platform: nordicnrf52; board: nano33ble; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nano33ble.html
PLATFORM: Nordic nRF52 (8.1.0) > Arduino Nano 33 BLE
HARDWARE: NRF52840 64MHz, 256KB RAM, 960KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES:
 - framework-arduino-mbed 1.3.2
 - tool-sreccat 1.164.0 (1.64)
 - toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Library Manager: Installing arduino-libraries/Arduino_APDS9960 @ ^1.0.3
Library Manager: Arduino_APDS9960 @ 1.0.3 has been installed!
Library Manager: Installing git+https://github.com/dlktdr/ArduinoBLE.git
git version 2.16.2.windows.1
Cloning into 'C:\Users\tobia\.platformio\.cache\tmp\pkg-installing-upn2fazb'...
remote: Repository not found.
fatal: repository 'https://github.com/dlktdr/ArduinoBLE.git/' not found`

https://github.com/dlktdr/ArduinoBLE.git is listed in platformio.ini as a dependency...
`
[...]
[env:HeadTracker_Nano33BLE]
platform = nordicnrf52
board = nano33ble
framework = arduino
lib_ldf_mode = off
lib_deps =
    Wire
	arduino-libraries/Arduino_APDS9960@^1.0.3
    https://github.com/dlktdr/ArduinoBLE.git
	bblanchon/ArduinoJson
debug_tool = jlink
[...]
`

I get that the non-zephyr-version might be deprecated, but why is it there then?

dlktdr, 2021-06-29:
The custom Arduino BLE code I actually just removed from my GitHub the other day starting to clean things up. And hadn't yet removed it from main. Didn't think about that my bad. I'll find an copy of it and put it back up there.

I would highly suggest using the zephyr version though. 1.0 was pretty buggy

omegat, 2021-06-29:
No worries, sounds great! :)
In truth I'm after the code to send trainer signals to a FrSky Para Radio. I used a PPM receiver until now but the new release of OTX broke the trainer port for me. So forking/building a PPM/SBUS->Para adapter on a basis of your code seemed like a good idea.
But I also have/use an instance of the head traker and really love it!

dlktdr, 2021-06-29:
Yuri was the one who initially figured out how to get the para working on the Arduino BLE library.

My modified library was the items he changed here and a few other minor ones that cause crashes at high CPU usage. https://github.com/ysoldak/HeadTracker

There should be I pre-modified one saved on my server still I'll upload when I get home.

dlktdr, 2021-06-29:
Also can't you just use my code as is and not have to modify anything?  It already supports PPM in and Bluetooth out?  Just leave all the other channels set off.

I mean I get the point of playing around with it too. Can just be fun.

dlktdr, 2021-06-30:
It's back. Old versions also removed from master.

Please see tag 1.0RC1 for the Arduino/MBED Code
https://github.com/dlktdr/HeadTracker/tree/V1.0RC1

omegat, 2021-06-30:
Good morning ;)
Thank you!! 
The initial plan was to remove all the AHRS-stuff and port it to a more cheap NRF52832 module. But you certainly have a point, especially because the tracker is not used 99% of me flying... I'll test that for sure! 
Thanks again and keep up the great work!

dlktdr, 2021-06-30:
I guess the other reason I would suggest Zephyr is how sketchy I found the ArduinoBLE lib at times.. but without all the extra CPU used you should be okay.

FYI. Something I found later that's not in that code. Increase the BLE connection interval in Arduino/MBED to min 6 (7.5ms) max 10 (12.5ms) & increase the update interval more. Will help the get those updates to the radio quicker.

closed 2021-06-30 by dlktdr