headtracker/HeadTracker #50: PPM Trainer Input NANO 33 BLE Master

Chefhandwerker, 2021-07-20:
I have a Nano 33 BLE on the google and transmit channels 6-8 with Bluetooth to a Nano 33 BLE Master on a TX16S.
A PPM receiver for trainer operation is still connected to PIN 9 on the master. As long as the Bluetooth transmitter on the google is off, the trainer operation works with the PPM receiver on the master. As soon as bluetooth is on, the trainer is no longer operated via pin 9

dlktdr, 2021-07-20:
If your on v2 there is an outstanding issue i haven't fixed yet. #39 

Where is you turn on the BT it overrides all BT channels over the PPM ones.

Haven't had a chance to get to it yet. No one had asked yet so wasn't on the top of my priority list. Is this what your seeing?

Chefhandwerker, 2021-07-20:
Hello, yes, I'm on V2.
That would be great. With the V0.xx I don't get the GUI to run after flashing on Para Master. With the V2 everything works, except on the PPM input in connection with Bluetoohs

dlktdr, 2021-07-20:
The para master fw didn't connect to the GUI in <1.0 It just looked for the first connection and started out putting data was pretty basic.

Was having a heck of a time getting it to detect the extra ble data sent in the new zephyr os so I put it on the back burner.

I'll give it a go tonight but what I can probably do is just give you a custom one where only Bluetooth channels 6-8 are overwritten so at least you can go fly :) That's an easy change for now.

Chefhandwerker, 2021-07-20:
That sounds great
I use channels 1-4 at the PPM input of the master and channels 6-8 from google via bluetooth.

dlktdr, 2021-07-21:
Here is a temporary binary you can use.

Ch1-4 PPM 
Ch 5-8 BT
Ch 9+ PPM

[BLE v2.03_BTCh5-8.bin.zip](https://github.com/dlktdr/HeadTracker/files/6852103/BLE.v2.03_BTCh5-8.bin.zip)

I'll work on getting this finished properly but this should work for you for now.

Chefhandwerker, 2021-07-21:
Hello,
many thanks. It works now.

dlktdr, 2021-11-05:
Fixed this issue now. Will either backport it in 2.05 or release 2.10 soon

dlktdr, 2021-11-07:
Version 2.10 posted with this fix.

Please re-open if the problem persists

closed 2021-11-07 by dlktdr