headtracker/HeadTracker #52: Smart Port Output on Master

dlktdr, 2021-07-23:
Moved From a Pull Request #51  to an issue

Originally posted from @Chefhandwerker

Hello,
is it possible to output the signals in the master as a smart port sensor?
The problem is, the FrSky X-Lite has no PPM or SBUS input. Just one smart port. This would make it possible to use the head tracker on an X-Lite.

dlktdr, 2021-07-23:
Can you install the Blutooth module in it then it will work. Here is a video https://www.youtube.com/watch?v=fyT6XohZIsY

I don't know if I can make the smart port send channels to a radio. I haven't really looked into it. I thought it was more telemetry related. I 'll do some reading when I get a chance.

Chefhandwerker, 2021-07-23:
Soldering the bluetooth module in the X-Lite is an option.
With SmartPort you don't have to output PPM impluse. Just a sensor. E.g. voltage 0-10V. You can then process that further with mixers

Chefhandwerker, 2021-07-23:
But the nice thing about an external BLE33 master module is. That I can use head trackers and teachers / students at the same time

dlktdr, 2021-07-23:
Yeah I guess that makes sense taking the telemetry signals and then mixing them into a channel 👍  I'll see if I can find some details on the electrical protocol if it's something the NRF52840 can do first of all.

dlktdr, 2021-07-23:
On first look the Half Duplex of it is going to be an issue to do all in the NRF. I still haven't been able to get inverted SBUS receive working. SPort looks inverted too. I can invert the signal but still haven't been able to get UART connected to that pin without it disconnecting my inverted signal. 

Would require switching back and forth from receive to transmit mode which might be do-able if the NRF will quickly allow pin switching without disconnecting my inverted connections.

If I can get inverted SBUS RX working is going to be step one then this might be possible. But don't hold your breath :)

dlktdr, 2021-07-23:
Might be able to use something like this to do the Half Duplex part.. https://www.frsky-rc.com/product/sp2uartconverter/

olexs, 2021-07-23:
I did something similar a while back, using a Teensy board to adapt a CPPM signal from a Fatshark goggles headtracker to S.Port for the X-Lite servo connector. I used the S.Port library from here: https://www.rcgroups.com/forums/showthread.php?2245978-FrSky-S-Port-telemetry-library-easy-to-use-and-configurable

My code is here: https://gist.github.com/olexs/955676a876c42fd1bd8cf33756d01cf0. It sort of worked :) although I never went beyond a proof of concept stage with it really.

Hope this helps.

dlktdr, 2021-07-24:
@olexs Good info!

I took a quick look to see if you can make the NRF52840 multidrop UART. Looks like it's possible to still have the TX pin defined as open collector and hooked up to the UARTE peripheral. You could bridge then solder the RX and TX together to get this functionality and just disable Rx when sending data.

If I can just get that RX and PPI channel working it might be fun to try.

dlktdr, 2021-07-28:
Found a solution to SBUS input inversion.. Not like I was hoping for but the user is going to have to solder two pins together. decided on pins D5+D6 as I haven't seen any photos or builds yet that have been using these pins.

A step closer to giving this a try.