headtracker/HeadTracker #53: very short range

twistedwings, 2021-07-30:
The range seems to be very limited.
In a distance of more than 2 meters I get
"Trainer signal lost / Trainer signal recovered" messages.

Arduino Nano 33 BLE
FW 2.0.3
RadioMaster TX16S
OpenTX 2.3.14
Para BT module

dlktdr, 2021-07-30:
Please check issue #48. Possible bad antenna on board. Also check antenna in radio too.

twistedwings, 2021-07-30:
thank you for the advice.

The capacitors on the arduino seem to be ok.
I cannot see any hardware issues on the para module.

I scanned both arduino and radio with "nRF connect".

Arduino distance 20cm
[image]
Arduino distance 5m
[image]
TX16S distance 20cm
[image]
TX16S distance 5m
[image]

Para module seems to have a stronger signal.
Do you have any rssi reference values ?

dlktdr, 2021-07-30:
I can copy them here when I get home from work tonight but if you sign into the discord chat and scroll up somebody else had the same issue where I took some readings at various distances

dlktdr, 2021-07-31:
Posted Db results I found in issue #48. Unfortunately I've already turned up the Nano's tx power as high as it can go. Usually 2m should be enough for a head-tracker but if you need to go further might have to use PPM cable. SBUS over a long cable length isn't a great option either.

twistedwings, 2021-08-01:
thank you!
Your values are similar to mine.
So rssi values and range seem to be normal.
I wonder about the range because BLE range is specified with 10 m.
Also 2 m range for the normal wireless trainer function (buddy box) I would consider as very low.

dlktdr, 2021-08-01:
Totally agree and the PCB antenna on the Nano isn't as good as it could be, wish there was an external antenna or I could crank up the power more. I'm not sure why FrSky opted to use such a limited wireless technology for this application to start with. Bluetooth **Low Energy** is just a bad idea to start with IMO... Although it did allow easy use with this board as a Headtracker, and generally your well within 1m.. so that's a plus 👍

Had it been me designing it I would have added a Frsky receiver (D8/16) in the radio for trainer like Specktrum does. I'm sure there was a reason, I just don't get it. Would have had range and low latency with a chip you already use?

I am still not getting trainer lost signals quite as frequent as your mentioning at 2m. It gets sketchy at around 5m for me. Although I have the PARA module with external antenna in my TX16s... probably helps. (Can't buy it I don't think. It came out of my Taranis X9D +2019.)

twistedwings, 2021-08-01:
D'accord.
Seems like we have to live with it.
In a distance of about 1 m I got 3 trainer signal lost messages in an hour.
For the headtracker that's acceptable.

closed 2021-08-01 by twistedwings

dlktdr, 2021-08-01:
1m is an awfully short distance to be loosing signal still. If you ever come across someone with another Nano would be worth a try or another with a PARA module to buddy box with to compare range.
I haven't re-checked it with 2.03 but I didn't change any BLE code so I don't think I botched anything up :)