headtracker/HeadTracker #54: Headtracker Windows APP on Android device to easy change settings at the field

maRCnh33, 2021-08-08:
Hello, I Just installed the headtracker on my plane and it works! But now i maybe have to change some settings when i am testing the new setup with headtracker and pan/tilt. 

Now i have only the option to take my laptop with me to the field...

But.... I would be VERY VERY nice if there is an android version of this app.

Can you do this? Can someone maybe convert it to an android app??

I have searched the internet and there is an option to convert windows app to android apps but it work only with a setup file, and there is no setup file, only EXE....

Hopefully it is possible to do this...

Tnx

dlktdr, 2021-08-08:
Since the GUI is written in Qt it should compile for Android too. There are probably going to be some issues but the biggest issue i can see is the layout of the GUI is too big for a phone as is. Tablet might look okay.

I don't have the Android tool chain setup on my PC anymore. If anyone out there would like this too

dlktdr, 2021-08-24:
Apparently you have to root your phone to get android USB serial port working 👎

Which means a big code change to use over BT. don't expect this one anytime soon.

rotorman, 2021-08-24:
Dunno if this helps, but [Mission Planner](https://play.google.com/store/apps/details?id=com.michaeloborne.MissionPlanner&hl=en&gl=US) is running on Android, and is able to bidirectionally communicate via CDC/VCP without requiring rooting (I am using it to connect to TX16S top USB-C port, where it gets sent MAVLink data from MAVLink router running in EdgeTX).
https://github.com/ArduPilot/MissionPlanner

dlktdr, 2021-08-24:
Mission planner on android now!! Love it.

Was doing some reading on QSerialPort in QT and there were lots of comments about CDC not working. Didn't even think to see how this was done here. I guess I should look at QGroundControl too, since it's the same toolkit too if they are using CDC or something else.

Thanks!

rotorman, 2021-08-24:
QGroundControl is definitely a good idea, yeas.

Also found following German language article from a user having the same root requirement issue with Qt initially, but he was able to solve it. The example code is likely the interesting part, possibly google.translate gives the gist of the text as well:
http://digital-nw.de/Seriellport-unter-Android-mit-Qt.htm

dlktdr, 2021-08-25:
Well that compiled a lot easier than I thought it was going to go. Last time I tried QT with android the setup was a b.... wasn't too bad at all in the current state. 

Will have to try on my actual phone to see what CDC does. I see lots of screen size issues that need to be sorted too.

[image]

dlktdr, 2021-08-31:
**Update:** After browsing and trying out code from QGroundcontrol I'm pretty sure this is going to work just fine without having to go directly via BT or any rooting.

Looks like they have used https://github.com/mik3y/usb-serial-for-android serial for android lib, then made a QSerialPort wrapper with JNI to it. There is only one function I used in QSerialport that wasn't implemented, and can probably get around it. Now I just have to learn some Java & Android to get past these last compiling issues. Also very similar to what the above German code did.

Also started a Separate UI just for Android, will have to separate everything into a few tabs so it's easily visible and hopefully won't have to make many code changes otherwise.

dlktdr, 2022-01-26:
Was looking at this idea a little while ago and started working on a JSON based BLE interface to the current serial data. Thinking about using WebBLE and (slowly...) working towards using a web based interface which is true multiplatform.

I think this is a better approach since I would also like to work towards a ESP32 based board and use Wifi/BLE/USB to configure it without worry about what OS the user is running.

dlktdr, 2022-12-06:
I'm going to close this one out.

I have incorporated a WebBLE method of adjusting the most common settings via a phone in the field. I currently don't have any plans to continue working on the Android version of the code. The Serial interface was a bit of task to make it work and it would not have helped any IOS users. Please see the most recent developments about Web BLE on discord for more information.

Testers requested!