headtracker/HeadTracker #56: Cannot Get SBUS to work with TX16S UART

robmitch111, 2021-08-19:
I've tried both single wired and remote BLE unit SBUS to AUX1 on TX16S with no luck. Used both BLE Tx and Rx pins as well. Used Hardware set to Serial Port SBUS Trainer, Model Trainer set to Master/Serial. 

There was talk of hardware inverters needed etc. Any suggestions? 
BTW, there is only one serial port option, not 2 in my TX16S radio.
PPM Master/Jack works fine, would just prefer SBUS
Using headtracker FW V2.03 (didnt work on V2.0, and PPM was jittery there too)

dlktdr, 2021-08-19:
Just recently found out the radiomaster needs open tx firmware of 2.3.10 or newer. They made a change here that allowed the use of the Uart as Sbus.

What version is on your radio.

It should be on pin Tx on Nano

dlktdr, 2021-08-22:
Going to close this out. Let me know if that didn't work, will re-open the issue.

closed 2021-08-22 by dlktdr

robmitch111, 2021-08-23:
Just did it. Yes, it works. Thanks dlktdr