headtracker/HeadTracker #57: Headtracker not working with Spektrum radio

QUADMAN7, 2021-08-23:
Hi,
I have made the headtracker and setup my spektrum DX6 gen3 according to online videos about setting headtrackers on spektrum radios however, when i plug the headtracker into the trainer port in the back  of the radio and move it around, none of my gimbal servo's move. I am using a 2s LiPo to power the headtracker and I have checked if it works in the GUI and it does as expected. Any suggestions would be greatly appreciated. THX!

dlktdr, 2021-08-24:
I guess I would be looking carefully at the wiring. Make sure PPM out pin is selected correctly in the GUI for the pin you have it wired to (D10 usually) and connected to the tip. If your using a stereo cable I know on other radios it didn't work great & needed a mono cable to get a proper connection. 

I don't think Spektrum uses an inverted PPM signal but can try clicking invert in the GUI to see if that makes a difference.

Sold my DX6 before I made this so I can't give it a try here. Double check your trainer mode is set as Programmable Master. If you have a trainer switch set is it actually on?

QUADMAN7, 2021-08-24:
Thank you so much for the suggestions! On my DX6 I set the trainer mode to FPV instead of programmable master so that may be the problem. Thanks a lot! 😁👍

QUADMAN7, 2021-08-24:
Hello again, I have tried everything you have suggested apart from using a mono cable since I don't have one at hand but I have tried several stereo jack cables. I checked my wiring, everything is all good ( if it helps, I have mine wired up exactly like Painless360 using the same parts) and yes, I have flicked the switch on my DX6. I also changed the trainer mode from 'FPV' to 'Programmable master' and experimented with inverting the PPM output but still nothing happens. Basically, all I know is that the head tracker works because all the channels move in the GUI and the centering button works, I know my DX6 recognizes it as a slave because it doesn't say 'NO SLAVE!', which must also mean the cables I have been using work but I am still stuck trying to get this to work, although I don't know if just because my cable is transferring enough DATA for the DX6 to recognize there is indeed a slave plugged into it it doesn't mean it is transferring all the DATA. Do you know if there is anything else I can try or if I just have to get a new radio that runs OpenTX like a TX16S? Thanks again.

dlktdr, 2021-08-24:
I feel like there should be a way to make it work. Have you tried sending the data on other channels, to see if you can at least get it to work as a normal trainer at least.. Setting the tilt roll channels to 1 2 in the GUI. See if it overrides your ailerons & elevator when you flip the switch?

You might want to try posting this to RCgroups fourm too https://www.rcgroups.com/forums/showthread.php?3822641-FPV-Head-Tracker-Single-board-wireless-easy someone there might have already done it on a Spektrum.

QUADMAN7, 2021-08-25:
Thank you for your help. I updated my DX6's firmware and everything is working now. I'm not quite sure what was causing the issue but it is solved and I have an awesome head tracker that works. thanks for all your help though. :)

closed 2021-08-25 by QUADMAN7

dlktdr, 2021-08-25:
Good to hear!