headtracker/HeadTracker #58: Pause Mode / Disable HT

robthomson, 2021-08-24:
I kind of feel you should consider a setup similar to to what the old trinity head tracker does.

On power up - send only a 'center' signal to the radio.

A long press of the button then turns the HT on.  

Conversely.. if flying; you can long press the button to 'hard center' the head tracker.

This makes it really easy to turn the HT on / off if you are experiencing issues mid flight!

dlktdr, 2021-08-24:
Not a bad idea. but if the board is screwing up I feel like having it output no pulses at all would be a better option, let the radio see the loss of trainer and it re-center. If can't trust the Tilt/Roll/Pan output then probably can't trust the center signal being sent either.

You have long press listed twice I would have thought another long press will turn if off again. 

Can you not accomplish the same thing on your radio with an override switch. Flip it on & forces your channels to center? I know I mentioned that on RC groups forum but I don't know if I actually put that in the docs. (I'll be sure to add it). Highly recommended! Also that way it won't matter what signal is being sent you can always force it center. Not sure if the Jeti has this capability tho.

robthomson, 2021-08-24:
Hey

Totally possible to do on the radio.  I do actually do this too!

Mainly it means that if you power up you get a consistent and stable output so the HT is not banging servos around until you are ready.

No pulses would likely work - however you would probably get allot of 'lag' before the radio decided to recenter!

dlktdr, 2021-08-24:
Ahh for that reason makes a lot more sense! I can add a checkbox beside the center button pin choice, so as not to screw up everyone that expects it to auto turn on.

dlktdr, 2021-09-08:
To be tested still but initial modification done. 

https://github.com/dlktdr/HeadTracker/commit/38d79c3b03ab20fc592bb8c9643ce25bf13a69d1

robthomson, 2021-09-08:
Nice!!

dlktdr, 2021-09-29:
Added the option to the GUI, and tested. Appears to work well. Defaults to off as not to affect users who are expecting it to turn on automatically.

While using this with the GUI I foresee an issue with users clicking this box, the Tilt/Roll/Pan graph still shows movement even though the actual outputs channel monitor is fixed to center. I've added a callback so the GUI can grey out the tilt/roll/pan outputs while disabled although not yet implemented. 

[image]

If you want to give it a try I would like to hear your feedback. You will need to compile the code in master. OR can upload the .bin and replace headtracker.exe with the one from 2.0 in this zip.

[HTLngTchEnaDis.zip](https://github.com/dlktdr/HeadTracker/files/7250132/HTLngTchEnaDis.zip)

dlktdr, 2021-11-07:
Posted in Version 2.10

closed 2021-11-07 by dlktdr