headtracker/HeadTracker #60: Erratic Servo on BT

robmitch111, 2021-08-30:
Hi, whilst I was able to setup the Head/Receiver pair over BT ad get it to talk to my TX16S with SBUS, I am getting very granular and delayed servo movements (just like the video posted in https://github.com/dlktdr/HeadTracker/issues/29)

I just put out the PWM for my pan channel to output A0 on the head unit, and the servo movement is silky smooth. However, when I go to BT, the servo output is erratic. Using FW 2.03
Cheers,
Rob

dlktdr, 2021-08-31:
The BT is a little bit jittery still. I've had issues getting the update rate fast enough that there is not noticeable steps between the updates but wasn't able to get it too much better than this.

Back in v0.8-1.0 it was very bad, was able to increase it quite a bit when shifting over to zephyr os for 2.0, In 0.9 I added a little filtering on the receiver board to smooth out the jitter which did help quite a bit. I could add an option to do this again, now I don't think it would require as high of a beta value on the low pass too.

dlktdr, 2021-09-01:
I was going over trying to get the Bluetooth output and didn't even think to mention if you have a bad connection the output is extremely jittery. If I place my hand over the antenna I get what you are describing too. Double check both boards antennas are intact #48.

I just posted V2.04. I didn't notice but the TX power wasn't set to max before, I had it commented out.. oops. Give it a try and see if there is any difference.

dlktdr, 2021-11-07:
Can you please test again on Version 2.10. Arduino to Arduino updates are much smoother. SBus output also now able to be set to up to 150hz.

Ideally in this case set it to 80hz. Which is the same as the Bluetooth update frequency now

Please re-open issue if required.

closed 2021-11-07 by dlktdr