headtracker/HeadTracker #61: Incorrect Output on Tilt/Roll/Pan Graph

dlktdr, 2021-09-05:
See

https://www.rcgroups.com/forums/showpost.php?p=47761965&postcount=435

steph280, 2021-09-05:
Configuration file in use
[SCSetting1.zip](https://github.com/dlktdr/HeadTracker/files/7111914/SCSetting1.zip)

steph280, 2021-09-05:
Original config file.
[DaveSetting1.zip](https://github.com/dlktdr/HeadTracker/files/7111930/DaveSetting1.zip)

dlktdr, 2021-09-05:
Hmm I don't see too much in there that should be causing an issue. Only thing I noticed is,

```
gyrxoff=-1.09
gyryoff=0
gyrzoff=0
```

Very strange for a perfect zero value on the y + z gyro offset values. When you calibrate on the first page are y + z raw values changing as you rotate the board?

[image]

steph280, 2021-09-05:
Yes it does.  The last settings I uploaded wasn't calibrated on this board.  Attached is the one after calibration.
Yes 2.0 GUI.
[SCSetting2.zip](https://github.com/dlktdr/HeadTracker/files/7112092/SCSetting2.zip)

dlktdr, 2021-09-06:
Do the gyro values seem close to the correct values?  I know its really hard to tell from the GUI but if you rotate the board 180 deg over 2 sec should see apx 90deg/sec. Just make sure it's not an order of magnitude off or something. The magdjwick filter used here is highly dependant on the gyro, then accelerometer to correct for drift in tilt/roll and finally the mag to stop pan drift.

Im guessing that there is some issue with the IMU on there. I'll make you up a firmware you can load on there to check the output of the IMU we can compare against.

I see @rotorman  got you to tey

dlktdr, 2021-09-06:
[v2.04_show_accgyrmag.bin.zip](https://github.com/dlktdr/HeadTracker/files/7117327/v2.04_show_accgyrmag.bin.zip)

This one will output all the sensor data.

Note the GUI is pretty slow with this one. But you should see acc(x,y,z) mag(x,y,z) gyro(x,y,z) in the live data. 
Accelerometer should +/- 1 in all directions. Mag for me (where I live) has the highest value in z, when board flat (-97). 

[image]
[image]

steph280, 2021-09-07:
This sounds silly but the issue turned out to be a constant short on the switch for center.  Re-solder to a different leg on the button and issue is resolved.  Thanks again Cliff!

dlktdr, 2021-09-07:
Lol. all good. Well this will be here for the next time someone does it.

closed 2021-09-07 by dlktdr