headtracker/HeadTracker #62: Pan drift after a quick movement

a1axx, 2021-09-15:
Hi, 

Recording of the pan drifts seen after a fast movement.

It does settle down after not moving for 30 seconds, but after another movement, it does start to drift again.

https://www.youtube.com/watch?v=FGBcDzp-dWk

a1axx, 2021-09-15:
More erratic in calibration of the Mag & 15% on low pass filters seems to mitigate the symptoms quite a bit. Some drift, but recovers faster.

dlktdr, 2021-09-15:
That's almost always from a bad Magnetic calibration. (Red sphere part). If I do the calibration near my PC speakers I get the same thing. 

https://youtu.be/tFwxMSW4uI0

Here is a video of mine after, see the same thing, Just drifts like crazy. Also showed a complete calibration working you can compare to. (Save to NVM and do a reboot after, as it can take a little while after calibration to stabilize if your last one was poor) On startup it takes a rough guess at which way it's pointing to reduce this time.

Best thing to try first is to get a decently long USB cord, go outside and re-calibrate with attached to a laptop. Do it the first time with the board only no - goggles. Then again with the goggles and powered up and see it that's where the interference is coming from. On my EV200 goggles I couldn't even use it if I had the board at the front.

We just need to make the Earths magnetic field stronger so this isn't so touchy :)

rotorman, 2021-09-15:
@dlktdr The video link in your last post on YouTube won't play - YT says the video is private.

@a1axx It also makes a difference if you calibrate indoors or outdoors according to my experience. Best result is outdoors somewhere in the field with nothing but grass around...

dlktdr, 2021-09-15:
Should work now.

a1axx, 2021-09-15:
Thanks guys, this has definitely been fixed with a more thorough mag cal and faster movements during the cal.

closed 2021-09-15 by a1axx