headtracker/HeadTracker #65: Firmware Not Starting Correctly On Power Up

umdpru, 2021-11-06:
I have flashed FW v0.92. Occassionally, the arduino board boots up into the blinky program instead of the headtracker firmware. When I hit the reset button on the Nano it reboots into the headtracker firmware.

dlktdr, 2021-11-06:
Highly suggested to use 2.0 instead. 0.92 is pretty old. lots of fixes since then.

https://github.com/dlktdr/HeadTracker/releases/tag/v2.0

dlktdr, 2021-11-06:
Just another thought too if your power flicker on startup can't be mended with the power connector you could also add a power up delay circuit to the board with a resistor and capacitor that should prevent any flicker from entering bootloader mode

[image]

umdpru, 2021-11-06:
Thank you. I will give the V2.0 a shot!

umdpru, 2021-11-06:
just flashed that version and did the setup. the power flicker problem persists and this is using USB to power it. I will wire it and check how it performs when powered direct via Vin. Also, V.92 was used following the videos on youtube. They obviously need updating. I had problems with servo outputs not showing up on radio with V.92 also but all fixed with V2.0.

dlktdr, 2021-11-06:
Might want to try a different USB cable and power source too if your powering via USB. Could just be a bad cable causing a power issue.

Actually just posted 2.10 today too.. :) Keep an eye on https://github.com/dlktdr/HeadTracker/releases for updates.
There isn't a lot new on this one, few new features and a couple bug fixes.

If you need help setting up can also get support on Discord or RC groups. Links at the top of https://headtracker.gitbook.io/head-tracker/ docs page.

dlktdr, 2021-11-21:
@umdpru Were you able to solve your issue?

dlktdr, 2021-12-16:
Haven't heard anything about this in a while. If you need the issue re-opened please comment

closed 2021-12-16 by dlktdr