headtracker/HeadTracker #67: GUI Application Crashes when Trying to Calibrate

dlktdr, 2021-12-11:
Reported by user dmpriso on RGC. https://www.rcgroups.com/forums/showpost.php?p=48311805&postcount=555

If it's crashing as soon as you try to calibrate I assume it is some kind of issue with OpenGL and GLU on your system. Can you provide any other graphics information?

GLU isn't recommended for new designs, the calibration sphere which visualizes the calibration was used from another open source project. Looks like I might need to actually replace it with modern libraries.

If you could try **replacing** this Headtracker.exe in the folder with this one [HeadTracker.zip](https://github.com/dlktdr/HeadTracker/files/7697311/HeadTracker.zip) and trying again. All the GLU requirements were removed from it. You won't get the visualization but can at least see if this is the issue.

dmpriso, 2021-12-14:
Still crashes, unfortunately. https://calip.io/SEsJ7Fp5#rKdkV34q

dmpriso, 2021-12-14:
I've created two crashdump files, one of the regular v2.1 and one of the special version:
https://drive.google.com/file/d/18VsqHyaJkOtC64Acg2Yk2YRP6N-Wy1jv/view?usp=sharing

dlktdr, 2021-12-15:
Thanks for those. I'll see if I can figure anything out from them when I get home tonight 👍

dlktdr, 2021-12-15:
If you have the time you could also try installing QT and running it in debug mode to see what line it's dying on. https://headtracker.gitbook.io/head-tracker/support/development/gui

dmpriso, 2021-12-16:
I can debug that, no problem, but can't promise to manage it before the xmas holidays.

ucakar, 2022-01-05:
Hi,

Is it possible to compile app to use it on mac OS?

dlktdr, 2022-01-26:
Just going through some of these issues. I haven't been able to reproduce this issue. Were you able to find a cause?

ror1948, 2022-03-18:
Using win10 Acer Laptop which has been updated to latest.
This is a new BLE right out of the box.

I am also getting crashes when I hit the calibrate button on V2.10 and V2.11 using GUI 2.11(zip) but GUI displays it is 2.10 ?
https://calip.io/w5R9HQvO#qGSN7cQ7

dlktdr, 2022-03-18:
There shouldn't be a GUI 2.11 zip out there only 2.1 . The 2.1 GUI will have both the 2.10 FW and the 2.11FW available on it.
GUIs should probably be labeled 2.1x as they will work with and 2.10-2.19 firmware.

I haven't been able to reproduce this issue on any of my PC's. can you upload the crashdump files like dprismo did above. Maybe there is something common between the two that might help. If you can compile the GUI and run in in debug mode it would help me find the cause. It should halt on the particular section of code that is causing the issue.

ror1948, 2022-03-18:
yes no 2.11 zip

does this help...crash did not produce dmp file
[headtracker crash on win10.txt](https://github.com/dlktdr/HeadTracker/files/8305421/headtracker.crash.on.win10.txt)

as to compiling I do not know how but i am willing to try.
just point me in right direction.
thanks

dlktdr, 2022-03-18:
https://headtracker.gitbook.io/head-tracker/support/development/gui

Here is how to compile it.

When I get home I'll take a look at it and see if that dump helps

ror1948, 2022-03-18:
ok thanks..doing it now

ror1948, 2022-03-18:
ok got it compiled with no errors
but how do i run to debug it

thanks

https://headtracker.gitbook.io/head-tracker/support/development/gui

dlktdr, 2022-03-18:
Just have to click the play icon bottom left the one with the bug on it.

That will start it in debug mode and you just also have to make sure it says debug not release when selecting

ror1948, 2022-03-18:
run and debug on left then green arrow at top.

popup looking for jlink emulator

dlktdr, 2022-03-18:
Run and debug was for compiling the firmware in vscode

Should be in qt creator for the GUI. 

I can help walk you through it in more detail over discord later. I'm still at work and won't be home for another 5hr tho, not sure what time zone you're in.

ror1948, 2022-03-19:
I am in Ottawa

- Ralph

dlktdr, 2022-03-20:
Just to update this issue thanks to @ror1948 we have found it appears to be an issue with the graphics drivers, still no solution as of now. Maybe try updating the drivers on the PC or re-installing if you experience the same issue.

closed 2022-08-03 by dlktdr