headtracker/HeadTracker #68: Firmware upload problems

gbruand, 2021-12-22:
Hello,
I bought a brand new nano 33 BLE and I am trying to upload the v2.1 firmware. During the upload the 3 first step seems to work propely but the 4th one taking a long time and then i recieve an error message telling me that the program has failed. Did I miss a step to do it properly ?

gbruand, 2021-12-22:
Nevermind I found the problem I just changed the USB Cable (maybe the good one was able to send data or at a higher bit rate...I don't know XD)

closed 2021-12-22 by gbruand

dlktdr, 2021-12-22:
No sounds like you did it correctly.  To clarify it's stuck on "Waiting for programming port"? 

What it does is automatically try to put the board into bootloader/programming mode. At this point (step 3) the device will disconnect from your computer and re-connect as a new COM port. Well it's supposed to :)

Can try a few things. Plug it in, check in device manager that you can see the com port in there. (which is probably working if it gets past step)

Then manually force the board into bootloader but double pressing the white button on it. The amber led should fade in and out. When it's in this mode you should see a new Com port number in device manager. Lets make sure that's happening and it's not a driver issue.

also what version of Windows are you running? 

If you do see a new com port you could try manually forcing it to bootloader for programming. Bring up the firmware loading window where it's waiting for com port plug it in and double tap the button right away

Lol... just saw the update.. Good to hear

anom2183u12, 2025-11-04:
Im having the same issue, bought 2 arduino 33 ble's but it happens with both.
it doesnt get stuk at the 4 part but immediatly when it starts on the 5th (uploading firmware) it fails with the message program failed.
Also it sais couldnt enter bootloader trying original port but it says that all te time. even tho it has a green light next to the 3rd part.

[image]

PS: I tried 3 diffrent usb cables (including brand new ones) and multiple diffrent usb ports on my pc
and I also tried with diffrent firmware versions including 2.21 / 2.13 / 2.12 and 2.11
and I tried to reinstall the drivers with zadig and later I thought I fucked that up so I uninstalled the drivers and reinstalled them with the windows driver thingy