headtracker/HeadTracker #70: SBUS Input

dlktdr, 2021-12-26:
Putting this here as a reminder I need to get this feature working!

DavidFpv, 2022-01-24:
@dlktdr 

Hello,
I would be happy if you could help me.
I want to transfer information from Arduino to the X9D remote and see this information on the remote within the channels.
I understand you did something similar but I can not understand how ...
If you can help me or give me direction I would be really happy

dlktdr, 2022-01-24:
Are you using a Nano33BLE board (NRF52840?) If you are you can copy the code from my source but it's pretty tied to the NRF52840 and Zephyr OS. 

I used the code from https://github.com/bolderflight/sbus You can see his credit in my source here https://github.com/dlktdr/HeadTracker/blob/2184a00d2462b52fd74bbf8d323059f2c99bcd36/firmware/src/src/SBUS/uarte_sbus.cpp#L330-L380

Basically just need to setup the serial port correctly which is 100,000 baud 8E2 for sbus. The inversion part can be tricky depending on what your hooking up. Pass the data through this function and output to the serial port... 

Hope that helps.. contact me on discord if you want to chat in more detail. https://discord.gg/ux5hEaNSPQ

dlktdr, 2022-05-30:
Added to Version 2.2. 
Bug still exists where some values are not processed. Seems like half. See issue #91 
Needs to be tested with other SBUS sources too.

closed 2022-05-30 by dlktdr