headtracker/HeadTracker #71: Not able to perform initial firmware load

tarp2699, 2021-12-31:
Hi,  
I'm attempting to flash the firmware on my new arduino 33 ble.  I'm running windows 11 64bit.  
On my PC when I plug in the device it is going to COM6,  if I double click the button it goes to COM 10.

When flashing the firmware, it will stall out looking for a com port.  If I unplug and re-plug it during the process it detects it and moves on.  It looks like it will work and then fails.  There was another issue someone posted that was similar, they attributed it to a bad USB cable, I have tried several and that is not the issue.
Here's the log from the firmware loader

Downloading https://raw.githubusercontent.com/dlktdr/HeadTracker/master/firmware/bin/BLE v2.10.bin
Downloaded firmware from the internet
Determining what type of file this is...
Firmware is for the Arduino Nano BLE 33 in bin format
Discovered a new port COM6
Discovered a new port COM10
Starting: bossac.exe --port=COM10 -e -w -R online.fmw
Starting bossac.exe --port=COM10 -e -w -R online.fmw
Erase flash
Flash erase failed
Programmer Exit Error(1) Unknown error
Killed programmer process!

Any ideas?

dlktdr, 2021-12-31:
Wonder if it's a Win11 thing. I don't have a copy of it yet.

Could you try writing it from the command line and see if maybe bossac.exe is giving some other error that the GUI isn't picking up.

If you shift & right click in the headtracker folder, choose `Open Powershell Here`

Double tap the white button so it's in bootloader mode (pulsing amber light) and verify it sees it in windows device manager as COM10.

enter the command `./bossac.exe --port=COM10 -e -w -R online.fmw`

online.fmw will be the last one it downloaded from the internet

Let me know what it says

tarp2699, 2022-01-02:
@dlktdr , I put it in bootloader mode, in device manager it showed up as com7 this time.  Here's my output.
PS G:\Downloads\HeadTrackerGUI_x64 - V2.1\HeadTrackerGUI_x64 - V2.1> ./bossac.exe --port=com7 -e -w -R online.fmw
Erase flash

Flash erase failed

dlktdr, 2022-01-02:
Well not much more helpful
How about seeing what `-i` (Info) gives. Here is what I see on mine

```
PS C:\HT\gui\bin\HeadTrackerGUI_x64 - V2.2> ./bossac.exe --port=COM5 -i
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
```

dlktdr, 2022-01-02:
Just want to double check too that you have the Nano 33 BLE and not the Nano 33 IOT. They use a different version of bossac and I see a couple posts that will give you a `Flash Erase Failed` message if trying to use the incorrect bossac on the incorrect board.

[image]

tarp2699, 2022-01-02:
Crap, I bought the IoT.  thanks for that catch.

dlktdr, 2022-01-02:
No problem.. and damn 👎  
Th only one I know about I have listed on the docs here. https://headtracker.gitbook.io/head-tracker/getting-started/hardware-required If you find more let us know.

I guess Newark is the only US source I know of right now.

Rotogeeks here in Canada has a few (this ship to the US as well).

Appears there is a drastic shortage of the IMU chips that are on that board. Pretty tough to find IMU chips for anything right now really. Most suppliers show 100's to 1000's on order but still a ways out for delivery.

closed 2022-01-26 by dlktdr