headtracker/HeadTracker #72: Is it possible to compile app to use it on mac OS?

dlktdr, 2022-01-05:
Hi,

Is it possible to compile app to use it on mac OS?

_Originally posted by @ucakar in https://github.com/dlktdr/HeadTracker/issues/67#issuecomment-1005584281_

dlktdr, 2022-01-05:
Moving this to a new issue to make it more visible.

It's written in QT and I tried to keep everything as cross platform as possible so yes it should compile (probably a few errors on opengl related items). I know the firmware uploader won't work because it's fixed right now to calling bossac.exe (The uploader software)

I don't own a mac to try it on but someone on discord yesterday just gave me an idea on how to make it work using github actions..

If you know how to compile software and fix minor issues yes should work if you want to try.

dlktdr, 2022-01-31:
It can be done.

https://discord.com/channels/827622724565467196/827622724565467199/937067815867912263

gcmcnutt, 2023-01-15:
Ok, in prep for macos support:
- i now have sensible actions in place that produce signed code
- we DO need to get bossac support working (namely signed apps)
- for above i can do similar with GH actions using similar strategy
This means mac user workflow will be:
- install head tracker DMG
- install Bossac DMG (the one signed by, say, me)
- then things should be fine.  i will work on this workflow to try an end to end test
- (later we can figure out how to have head tracker ALSO install bossac so is more of a one stop shop)