headtracker/HeadTracker #79: Bluetooth trainer issue

Iamyahodfpv, 2022-03-20:
Hello, i have the arduino board paired with my qx7s but cannot get any response from it. outputs are set up and still nothing

dlktdr, 2022-03-20:
Which firmware are you using? Firmware less than 2.0 wouldn't work with the CC2540 module in some radios. Will connect but never receive data.
You have the channel outputs in the GUI for the nano set to something other than off?
Do you hear "trainer signal recovered/lost" on the radio and lost when the Nano is turned on and off? If you hear this it's sending data and probably just a configuration issue somewhere Radio or Headtracker GUI.

Iamyahodfpv, 2022-03-20:
I am using .92 for the firmware. i couldn't get the newer firmware to flash

Iamyahodfpv, 2022-03-20:
i have the channel outputs to 7 and 8

Iamyahodfpv, 2022-03-20:
I tried going from .92 to 2.11 and it says programming failed

dlktdr, 2022-03-20:
Yeah you probably have a CC2540 bluetooth module, will need v2..

As soon as it says "waiting for programming port" try double tap the white button quickly and wait. This will force bootloader mode.

Iamyahodfpv, 2022-03-20:
I tried to double tap the white button as soon as it said that and it still says programming failed

dlktdr, 2022-03-20:
Try this, Use the 0.92 GUI. 
Download this file https://github.com/dlktdr/HeadTracker/blob/master/firmware/bin/BLE%20v2.11.bin
Choose upload firmware, choose a local file and try selecting this downloaded file. Double tap, Refresh the ports and try uploading.

If it uploads see if you can then use the 2.1GUI

Have you tried other USB cords too?

Iamyahodfpv, 2022-03-20:
that worked, what bluetooth setting do i need to use in the gui?

dlktdr, 2022-03-20:
You want "head mode"
https://headtracker.gitbook.io/head-tracker/settings/gui-settings/bluetooth

Iamyahodfpv, 2022-03-20:
And just set the receiver channels in the gui?

Iamyahodfpv, 2022-03-20:
I set the receiver channels and it works! Thank you for the help!

dlktdr, 2022-03-20:
Yes just choose the channels you want the Bluetooth data to be output on the main screen (tilt/roll/pan) https://headtracker.gitbook.io/head-tracker/settings/gui-settings/tilt-roll-pan
You can only use Ch1-8 over Bluetooth.

If you need more help setting up be sure to sign into the Discord chat too.

Great to hear!

Iamyahodfpv, 2022-03-21:
I have one last question. I have the head tracker all set up but if I tilt the goggles too far up the servo resets to the bottom. What could be causing that?

dlktdr, 2022-03-21:
There is only 180 degrees of tilt available. If you go over 90 up or down thinks its upside down and flips the output. A eurler angle thing.. If you have the board mounted in a not flat orientation, you're going to lose some of this tilt range when you center it.

Take a look at the video on rotation here If you set this right, there should be more than enough range without it rolling over.

https://headtracker.gitbook.io/head-tracker/settings/gui-settings/general

closed 2022-08-03 by dlktdr