headtracker/HeadTracker #87: PAN elegant

sgtan1912, 2022-04-13:
Hello, I would like to ask you about my head tracking module, which was found to move with TILT. I changed the GYRO WEIGHT ON PAN to 100, but there is a new problem, that is, PAN will continue to be slow and flowing. What may be the problem, please help me

dlktdr, 2022-04-13:
Gyro weight on pan? Is this the Headtracker built from here? https://www.rcgroups.com/forums/showthread.php?1677559-DIY-Headtracker-(Easy-build-No-drift-OpenSource) The GUI Looks like this.. https://www.rcgroups.com/forums/showatt.php?attachmentid=5087241&stc=1

This project doesn't support that board only the Nano33BLE. https://headtracker.gitbook.io/head-tracker/getting-started/hardware-required

But in general any drift in Pan is caused by a bad magnetometer reading, or bad magnetometer calibration. One of the reasons I started this project is because getting the proper modules for the other project could be difficult. I also had issues getting that one to work properly. Take a really close look at the Magnetometer chip you have. The code is for a HMC5883, but quite often a QMC5883 is sent. There is a line of code in a .h file you can change to switch between the sensors types. Only in the newer versions, available here. https://headtracker.gitbook.io/head-tracker/getting-started/hardware-required

sgtan1912, 2022-04-16:
> Gyro weight on pan? Is this the Headtracker built from here? https://www.rcgroups.com/forums/showthread.php?1677559-DIY-Headtracker-(Easy-build-No-drift-OpenSource) The GUI Looks like this.. https://www.rcgroups.com/forums/showatt.php?attachmentid=5087241&stc=1
> 
> This project doesn't support that board only the Nano33BLE. https://headtracker.gitbook.io/head-tracker/getting-started/hardware-required
> 
> But in general any drift in Pan is caused by a bad magnetometer reading, or bad magnetometer calibration. One of the reasons I started this project is because getting the proper modules for the other project could be difficult. I also had issues getting that one to work properly. Take a really close look at the Magnetometer chip you have. The code is for a HMC5883, but quite often a QMC5883 is sent. There is a line of code in a .h file you can change to switch between the sensors types. Only in the newer versions, available here. https://github.com/kniuk/DIY-Head-Tracker https://github.com/kniuk/DIY-Head-Tracker/blob/3bd87d91b06dd10f6f321cd155c3c0897efcdfdb/HeadTrackerFW/Config.h#L43-L47
> 
> Edit.. Also with gyro weight on pan set to 100 on that project doesn't use the magnetometer at all, only gyro. Which will always have some drift associated with it. this is why a magnetometer is used to prevent that.

closed 2022-04-16 by sgtan1912