headtracker/HeadTracker #89: Impossible to deactivate a channel using option 2/3.

bvanderhoek, 2022-05-28:
With firmware 2.11 and HT2.10 app, it seems impossible to turn off a channel, for a test i had X,Y,Z active on channels 7,6,5. I wanted to deactivate Y, roll in the AUX config tab, but OFF doesnt seem to be stored in NVR. after reboot last channel number is restored incorrectly.

bvanderhoek, 2022-05-28:
Note last resort is full chip erase and reinstall firmware 2.11

dlktdr, 2022-05-28:
There was a bug here. Thank you for bringing it to my attention.

https://github.com/dlktdr/HeadTracker/commit/3336760798b746e5777dc944e43c419dd1a3a057

You can use this beta copy of V2.2, there is a "Restore Defaults" option in the GUI under Diagnostics to wipe the firmware values to default. Note this requires that the 2.2 firmware is also on the device.

[HT22Beta.zip](https://github.com/dlktdr/HeadTracker/files/8791268/HT22Beta.zip)

Replace the .exe with V2.1.
Use upload from local file and choose the BLE v2.21.bin firmware

Note this is a Beta Version there are other issues still, it's not complete.

closed 2022-12-10 by dlktdr