headtracker/HeadTracker #92: Pan Drift even after good calibration

dlktdr, 2022-05-28:
Let me start out by saying that all Mag sensors are a pain... sorry. If you have ever tried to calibrate a drone with an onboard compass you will know what I mean.

If your mag is calibrated and in a good noise free low magnetic environment you can still see drift in the pan axis.

The issue stems from how it determines the initial orientation and the sensor noise. The mag sensor on the Nano33BLE is noisy. Without a lot of filtering you get this but it responds to a changing field quite quickly.

https://user-images.githubusercontent.com/281145/170827908-9438a9b0-3fdf-4264-a9f4-308b1976d886.mp4

There is a very low amount of mag correction applied on every loop so that this noise doesn't show up on the servos. The result of this is that there is a slow drift until it stabilizes. Since it determines the initial mag alignment on boot, if its not stable on powerup this time to stabilize can be long.
 
### Notes

- Pan can also drift due to the gyrometer, calibration of this has been moved to every bootup in V2.2. Gyro's are more temperature unstable, even with temp compensation on in the LSM9DS1 it still can show drift. 
- Placing the Nano33BLE chip close to some electronics will cause it to go crazy. Nothing can be done about this, it needs more separation distance. You will see drift no matter what you do, no matter how long you wait as it can't make heads or tails of which way your pointing. There is no mag sensor around that won't cause an issue here.
- Permanent magnets same issue as above.
- **ALWAYS** do a calibration with your board mounted in the final location, everything powered up including your goggles with radio links on. This is **VERY** important, if your having issues. If you can't get  a calibration to pass like this it will never work right no matter how long you let it settle. 

### Solutions
Version 2.2, gyro calibration is done on boot and should help here.
Version 2.2, the mag sensor will be filtered by itself alone from the gyr and acc. This allows for quicker response in the fusion algorithm. Seemed to have good results in the tests.

TimothyGold, 2022-06-30:
I noticed that when i boot after i put everything on my head and home still until calibration finishing the pan will hold pretty good. This is until i need to rotate to get better reception. Once i do this and re-center drift will start. It does seem to get better again if i am able to make center closer to boot direction. This sometimes is not possible. Is it possible there is something missing in the calculation for re-centering?

dlktdr, 2022-07-01:
I'm hoping the changes to be added to 2.2 will address this better, but still needs more testing. 

2.13 was just posted that resets the algorithm on every center. It might help.

dlktdr, 2022-07-04:
Was checking into it a bit more yesterday and found another issue that might be accounting for a bit of this. There may be an issue with the integration of the gyro signal on fast moves, probably not enough samples to get a great result. If you do a quick movement and stop, the gyro has overshot where is should be, which then due to the filtering slowly drifts back to the proper mag center.

Velvethsaw, 2022-08-05:
I am too experiencing drift after good calibration. Thank you for working on this project!

dlktdr, 2022-08-05:
Would you mind testing this version of the software and see if you notice any differences?.

Below is copied from RGC

Drift Issues
For everyone experiencing the drifting. Trying some new things that might help. Paul_BB was nice enough to send over a new algorithm that is being tested. This coupled with 2.6x more sensor reads so far I think it's superior.

If you want to try yourselves, [https://drive.google.com/file/d/175v...ew?usp=sharing](https://drive.google.com/file/d/175vGJobJggTQzdDT1_VNp1uQdM2CUI3h/view?usp=sharing) - This hasn't been thoroughly tested. The firmware is included in the above file, under shipped firmware. I wouldn't trust it enough to fly with yet, just an initial test. Code is on Github under the Paul_DCM_AHRS branch

Note: Use the Diagnostics->Reset Defaults option from the GUI before starting.
Gyro calibration has been removed so it will take a little while for the algorithm to cancel out the Gyro drift. Like the current firmware please make sure to do a Calibration with headset attached and powered. After power up, hold still for a while, do some motions, then reset the center after it settles. A new center reset was sent as well, but not implemented in the above code. Leave your feedback

Note: (From Paul_BB)
The board has its own orientation. Tilt roll pan (pitch roll yaw) of the board are relative to the horizontal component of the local Magnetic field and to the gravity.
The centre reset will rotate the orientation of the board to the convenience of the user, when implemented.
There is another thing that should be implemented: the initialisation of the board pan at start up. Today the pan is set to zero (local Magnetic North) at start up. So if the board does not face North at start up, pan will need some time to catch up. This will be corrected (if the algorithm is promising).