headtracker/HeadTracker #99: PPM input not working

kesheshyan, 2022-06-16:
Hey, I was trying to read the PPM input as described in the documentation but it never worked.

I've tried different Arduinos, with and without voltage dividers - no success. As input I used 100% correct PPM signal from my servo tester device that I know for sure works.

I went ahead and flashed a board with Aduino code I wrote that just reads PPM signal without changing anything in terms of wiring and it worked immediately, so I assume the problem is related to PPM-decoder code.

dlktdr, 2022-06-23:
Did this solve your problem?

ldalba, 2023-05-21:
I have the same problem, no ppm input signal. I'm using the firmware 2.13v. Did you solve the problem?

dlktdr, 2023-05-21:
You wouldn't happen to have an oscilloscope you can post a photo of the PPM stream your sending into it would you?

ldalba, 2023-05-21:
> You wouldn't happen to have an oscilloscope you can post a photo of the PPM stream your sending into it would you?

I don't have an oscilloscope. I can read the PPM signal with the function "measure signal - PPM". 
For the wirings on the Arduino NANO 33 BLE, "PPM" is on D9 and "GND" on GND, as in the instructions.
By the software side (HeadTracker), on the channel viewer: NO PPM INPUT channels detected
[image]
[image]

dlktdr, 2023-05-21:
Did you choose the PPM pin in the GUI as well?
[image]

ldalba, 2023-05-21:
> Did you choose the PPM pin in the GUI as well? Probably should have a * and note beside the images in the docs, as you can pick some of them at will, this one defaults to "Disabled"
> 
> [image]

Of course. It was the first thing I did

dlktdr, 2023-05-21:
Just checking :) Without seeing what the PPM frame looks like I'm just guessing here but the Frame Sync Pulse is really short on whatever is generating it. 

I've reduced the minimum sync time to 2ms here and the min channel count to 2.

Try this, it should be compiling some output you can test to see if it works. Maybe take a look in the serial monitor as well to see if you get the PPM signal detected at all when attached.

Github Actions should be compiling the test firmware here,
https://github.com/dlktdr/HeadTracker/actions/runs/5040070851

Be sure to use a 2.2Gui as well. Can read how to get a development GUI about actions here if your unsure, https://headtracker.gitbook.io/head-tracker/support/development/git-workflow-and-actions

or can download the 2.2pre-release to get a 2.2gui too,
https://discord.com/channels/827622724565467196/827622724565467199/1106857976536506438

ldalba, 2023-05-22:
> Just checking :) Without seeing what the PPM frame looks like I'm just guessing here but maybe the Frame Sync Pulse is really short on whatever is generating it.
> 
> I've reduced the minimum sync time to 2ms here and the min channel count to 2. [14d0b3a](https://github.com/dlktdr/HeadTracker/commit/14d0b3a855721b02cc3907ac4937d2f2481a3be1)
> 
> Try this, it should be compiling some output you can test to see if it works. Maybe take a look in the serial monitor as well to see if you get the PPM signal detected at all when attached.
> 
> Github Actions should be compiling the test firmware here, https://github.com/dlktdr/HeadTracker/actions/runs/5040070851
> 
> Be sure to use a 2.2Gui as well. Can read how to get a development GUI about actions here if your unsure, https://headtracker.gitbook.io/head-tracker/support/development/git-workflow-and-actions
> 
> or can download the 2.2pre-release to get a 2.2gui too, https://drive.google.com/file/d/1DVPoDp4ZXhoI9SxvCinXkAfCYj1iWv1L/view?usp=share_link

It works fine just using the 2.2 pre-release! thank you. Now I can use my joystick logitech extreme 3d (ppm signal output with an arduino uno programmed) to drive the ariplane and also the headtracker for fpv.

dlktdr, 2023-05-22:
Awesome. 🎉 That sounds like a cool project, do you have a link to the code for the joystick part somewhere. I know some users on discord were looking for the same thing. Should come join us and post some photos in #flightstick too. https://discord.com/channels/827622724565467196/856339077115543562 😄 

Just to clarify, did you try the firmware that came with 2.2-pre-release? Does that one work too? Do I need to worry about including this timing change? [14d0b3a](https://github.com/dlktdr/HeadTracker/commit/14d0b3a855721b02cc3907ac4937d2f2481a3be1)

closed 2023-05-23 by dlktdr