headtracker/HeadTracker #109: Allow use while in a Car / Plane or other rotating body

dlktdr, 2022-08-03:
This has come up a few times, people looking to use the Headtracker while they are moving in a car, plate, etc. Right now Pan would be unusable. When you turn the car to the left that your sitting it the Pan output is also going to change.  

Proposal is using a second device, which is sending the reference rotation. The actual output is based on the difference between the two. You would place the reference Nano on the cars dash, the one on your head would be the difference between them.

It should work in all orientations ideally, if your upside down in a plane for instance. Finding a method to determine if both boards are on the same plane is going to be the tricky part. Or leave it up to the User to require them to place them correctly at first.

pcats, 2022-12-10:
This method does indeed work.

Quite a few years ago I was working on a project using a USB IMU unit for virtual reality applications, which required being able to perform head tracking while walking around, and that is the method we used. We had one unit on the headset, and one on the body, attached on a small harness on one shoulder so they were nominally in the same plane. Using the difference between the two IMUs allowed us to extract the relative position of the head to the body, which could then be used in our application. On the whole it worked much better than we'd initially expected as it was such a simple solution!

I would suggest that the simplest method would be just to leave the positioning of the reference IMU up to the end user, at least to begin with. If both are zeroed at the same time, it might not actually need them to be mounted in identical planes, thinking about it. We didn't go to any extreme measures to ensure that the two units were absolutely synced like that and it still worked surprisingly well.

Possibly adding a method to allow resetting either unit to reset the other one at the same time over bluetooth would be a valid approach?

bryceartstew, 2022-12-14:
Wow, That sounds like a useful solution for several issues. I was oddly thinking about this today and that was the idea of a solution I thought could work. Would've loved to have seen that VR system running.

DoodlesMcNoodles, 2023-11-08:
I am working on a project that falls into this category, which is to control servo panning of a GoPro camera on my bicycle via head tracking, using 2 Arduino Nano 33 BLE boards.  The camera setup will be attached to the center of the handlebars (or maybe the front of the bike frame itself).  The camera will be rotated directly by the output shaft of the servo motor, and the receiver Nano will be part of that assembly.   The transmitter Nano will be attached to the top center of my bike helmet.  Both Nano boards will be in the same nominal Z orientation at all times, and the same X and Y orientations nearly all the time (except when I'm actively turning).  I will nearly always be moving in a nearly level plane, unless I fall off my bike.

I've already put together the basic camera/servo assembly and breadboarded both Nanos for initial testing.  I have downloaded the Headtracker firmware to both Nanos and put them in BT bridge mode.  Did not bother including the magnetometer.  It works great sitting at my desk!  But I have already assumed I will have some issues when I am able to mount everything and try riding.  I had thought that using references from both boards would be necessary.  Possibly the differential Z-axis angle between the 2 Nanos would be the value to drive the panning servo.   In general the largest non-Z gyro value (normally along X) should establish the current line of travel.

I have to say, the Headtracker software is quite excellent!  Before I found your project I was trying to figure out the coding from scratch, and had enough problems even trying to calibrate the IMU correctly that I was in despair.  You have done a great job with this!  Thanks so much for all your excellent work!

I am a retired mechanical engineer and have a lot of experience with mechanical design, and some with programming and circuitry.  And a fair amount of free time on my hands.  I have played around with basic Arduino sketches and am willing to put in some effort to help this moving-system concept come to fruition.   I think my case might be a good starter since it only involves one axis of rotation.

Here's an entry I just made in Discord about it, with a photo and video (I hope this link works...)

https://discord.com/channels/827622724565467196/827622724565467199/1171872604152397834