headtracker/HeadTracker #115: GUI App: Enabled and disabled buttons on the main screen have very little visual difference

kobelev-tech, 2022-08-10:
In fact, there are several problems with these buttons:

[image]

1. When the board is disconnected, only the connect button should be visible.
2. There should be about 4-8 pixels space between the icon and the text on the button.
3. The background of active buttons should be brighter (like 240-244). The background of disabled buttons is fine.

[image]

4. The icons of disabled buttons should be little bit lighter to match the color of disabled text. Changing the color of disabled text is also fine. The colors should have the same brightness.
5. Only one of the enabled buttons is colored. It might make sense to add some color to all of them. Or have them all grey.
6. Remove the "Send Changes" button. In any case user cannot press it. It is only making slight unclear blinking.
7. Remove the "Channels Viewer" button. This is not an action on the board like all other buttons.
8. Move "Reset Center" to the left.
9. Move Save Settings to the end.

---------------------------------------
Naming consistency:

[image]

Titles of the dialogs:

- Diagnostic Display
- Serial Information
- Channel Viewer

dlktdr, 2022-08-13:
1) Having it disappear makes no sense, I'll change it to stay the same button Connect / Disconnect.
2) Can add a space or two to separate these at the beginning.
3-4) This is based on your windows theme. Looks better on Win 11, IMO. **Anyone want to do a style sheet here**, there are no code changes required. Just editing the stylesheet.css
5) Anyone want to draw new icons. I did one in B&W, looked like crap so I went with the color variant. I'm no graphics artist :)
6) Done. This is from old code brought forward, it's not needed anymore.
7) This one I'm not going to do, it used to be hidden in diagnostics. There was request to make it more visible, no one knew it was there. I agree, it's a pretty important tool while setting everything up. Can shift it and calibrate to the bottom (spacing in between), as they have nothing to do with connection or settings.
8) Already done in PR #112

dlktdr, 2022-10-15:
Mostly completed in #112

closed 2022-10-15 by dlktdr