headtracker/HeadTracker #116: Centering button

AndersN76, 2022-08-12:
Is it possible to make an internal button in the transmitter act as a centering button instead of having an external one?

dlktdr, 2022-08-12:
No unfortunately not, it's hardwired to the chip reset. and can't override it.

kobelev-tech, 2022-08-12:
It may happen that Anders was speaking about some button in the RC transmitter. Anders, can you please clarify your request?

gcmcnutt, 2022-08-13:
Well, @dlktdr and I have discussed how to have bi-directional link communication between HT and Transmitter just for this.  So, there is some work to do to get this working on all the various link techs.

For me, I want to develop it for BlueTooth connection.  But others are certainly possible.

dlktdr, 2022-08-13:
@AndersN76 - Sorry, I see what you actually mean. 

Some working options are,

* Opening your radio and rewiring a switch. Then run wires to the Headtracker. (Works will all radios, but there is an extra wire required)
* Rewire or add a switch in the radio and hook it up to a receiving Nano 33 BLE inside the radio ([Option 3](https://headtracker.gitbook.io/head-tracker/getting-started/wiring)). Then wireless connection up the the Head Tracker. (If there is room inside) 

Works, but I would do a lot of ground testing before I flew with it.

* If your using a very specific list of Bluetooth, EdgeTX and a TX16S (Or other color screen radio) it can be done using an ESP module in your radio and EdgeTX. Still a ways off from production, but we would love the support from anyone capable. Links below.

Ideas, most of these require software changes on the radio. They will be limited to open source radio firmware. Both of these can monitor a channel on the Head Tracker to cause a reset.

* SBUS input & output while in trainer more on the AUX ports available on some radios
* PPM In & Out in Trainer mode at the same time. (Hardware limited)

Others ideas not radio dependent, 

* Shake your head quickly or look some direction quickly 3 times. Basically any sequence, then look where you want and it's reset.

The wireless module code from above,
https://github.com/dlktdr/BTWifiModule - Code and binaries here.
https://github.com/dlktdr/BTWifiModule/blob/8bea9db7534ea96a0483e6965628af2f1c132c39/src/terminal.c#L148-L154
& this patch to EdgeTx 2.7.1 
https://github.com/dlktdr/edgetx/tree/Bluetooth-Mods_27
https://github.com/dlktdr/edgetx/blob/2f4d390349f5d619b386751876f8160300b09d06/radio/src/bluetooth.cpp#L477-L484
[EdgeTX271_Blutooth_HTReset_BTWifiModule.zip](https://github.com/dlktdr/HeadTracker/files/9330759/EdgeTX271_Blutooth_HTReset_BTWifiModule.zip)

closed 2022-10-15 by dlktdr