headtracker/HeadTracker #118: Cannot Get SBUS or PPM to work with TX16S internal port

skunkwurk, 2022-09-26:
Hello,

I have one unit working fine from the goggles to the radio's input/jack using PPM, works fine.  But I'd like to go wireless, I bought a second unit and have them setup.  I'm using a channel in the 1-8 range, have it wired per the instructions but cannot get it working.  I have an MK II radio, this radio not longer has the internal pad to solder a bluetooth module, it now has a connector. I connected the second BLE to the pwr, tx, rx, & gnd ports.  The unit powers up and BT connects, I verified the mac addresses. I just cannot get the radio to recognize the wired unit.  There is no local or remote address under hardware.  I checked/ enabled bluetooh as a build option for the radio as well.  I have tried both PPM and SBUS outputs on the BLE connected to the radio.  I am now at a loss, no clue what else to try.

Thoughts?

[image]
[image]

dlktdr, 2022-09-26:
On the second board it only will output in PPM or SBUS. It cant simulate the bluetooth modules interface.

You have a few choices here,

1) You can leave it hooked up here (aux2), switch the firmware on the radio to a non-bluetooth one, then pick SBUS/Serial trainer on aux2. With the bluetooth firmware of edgetx you can't select an option on aux2, its fixed to Bluetooth.

2) wire the arduion tx  into aux1 rx and instead then pick SBUS/Serial trainer on aux1 keeping the same firmware.
 SBUS Only needs the Tx pin connected.

3) wire the ppm output pin into the jack at the top internally.

Both of the SBUS options need to have the Inverted checkbox clicked in the GUI. The Mk2 aux ports are standard uart, which is inverted SBUS.

skunkwurk, 2022-09-26:
Hello Cliff, I like option 1.  Just to make sure I understand, are you saying the port I have it connected at currently in the pic is AUX2?  All I need to do it revert back to the non BT code?

dlktdr, 2022-09-26:
Yes that's correct, then pick serial trainer instead on aux2.

skunkwurk, 2022-09-26:
Sweet!  Lets me try that now and I'll circle back.  thanks for the assit!

skunkwurk, 2022-09-27:
Still no luck, do I need to keep SBUS on the remote unit set to 80hz plus invert?  I have all channels set to off.  I connected both the TX and RX wires, do I need just the TX?  Will it matter is both are connected?  All else is set as per your suggestion.

dlktdr, 2022-09-27:
Only need the Arduino TX to the MK2 RX, but having the other hooked up shouldn't hurt anything.

The two blue lights are coming on on both boards (they are connecting together)?

All channels off on the receiver board is correct. Head board should be setup the same as you had it with PPM. Just turning on Head Bluetooth mode.

I can go attach to my MK2, and take some photos if that doesn't help.

skunkwurk, 2022-09-27:
Wait a minute, under channel viewer I can see channel 5 moving when I move the axis for the headunit, so the sbus signal is getting to the remote unit.  I am just not seeing it as an input in the radio.  Something I noticed is that I do not see the channel in mixes highlighted in yellow, like I would when I was using the input jack port.  When I would connect it, it would audibly say trainer connected, and they would turn yellow.

skunkwurk, 2022-09-27:
The remote unit also shows CH 5 under BT inputs in channel viewer.  So this looks to be a radio config issue.

dlktdr, 2022-09-27:
Sorry you're seeing the channel 5 moving in the radio channel viewer as well?

If you unplug that connector you should still hear it. Say trainer signal lost.

skunkwurk, 2022-09-27:
To confirm, under hardware I am set to use AUX2. set to SBUS Trainer. Then in MDL, I set the trainer mode to Master/Trainer.  Do I need to set anything else differently?  All else I have setup as I would using the input/jack setup.

dlktdr, 2022-09-27:
Under MDL should be Master/Serial

dlktdr, 2022-09-27:
I have to make some photos for the Mk2 anyway, I'll go do it now :)

skunkwurk, 2022-09-27:
I'm sorry, I meant to write Master/Serial under MDL.  All is set, still no input on channel monitor in the radio.

skunkwurk, 2022-09-27:
> Sorry you're seeing the channel 5 moving in the radio channel viewer as well?
> 
> If you unplug that connector you should still hear it. Say trainer signal lost.

No, not in the radio.  I see the channel signal in the remote BLE.  I can see CH 5 in the BT tab as input, and I see CH 5 as SBUS output.  But when I check channel monitor in the radio, I do not get any movement.  In fact, I do not hear trainer connected or signal lost when I disconnect it,  Seems like the issue is the port I'm using.

dlktdr, 2022-09-27:
I'll keep updating this as I go.

[image]
[image]

skunkwurk, 2022-09-27:
So far, we're wired the same.

skunkwurk, 2022-09-27:
I removed the SBUS RX, mine matches yours.

dlktdr, 2022-09-27:
Yep your right, this should be working, it's not. Pulling out the scope. What EdgeTX version are you running, mine was shipped with 2.6?

skunkwurk, 2022-09-27:
> Only need the Arduino TX to the MK2 RX, but having the other hooked up shouldn't hurt anything.
>
> The two blue lights are coming on on both boards (they are connecting together)?
>
> All channels off on [...]

yes, the blue led lights are on for both units.  I confirmed they are connected via the mac address of the headunit, plus the channel viewer of the remote units shows good communication.  my issue is the connection to the radio.  it is wired per your photos. 

> Yep your right, this should be working, it's not. Pulling out the scope. What EdgeTX version are you running, mine was shipped with 2.6?

I'm on 2.7.1, Black Pearl.

dlktdr, 2022-09-27:
Found it. The silkscreen is labeled wrong on the board. Move the wire to TX it works. 

[image]

skunkwurk, 2022-09-27:
You are the man, it works!  A couple question, any chance we could use higher than 8 channels?  I am currently testing using channel 5 but typically use channel 10 and 11 for pan/tilt.

dlktdr, 2022-09-27:
👍 Not over Bluetooth, limited to 8. But since it's Edge/OpenTX, the incoming channels it's on doesn't really matter you can set whatever input you want to whatever output. 

E.g. If you set your channels to Tilt/Roll/Pan to 1/2/3 respectively, on the head board. Then under Mixer channel 10, choose TR3 as the input. Mixer Channel 11 choose TR1 as the input.

skunkwurk, 2022-09-27:
nice!  I love EdgeTx.  I am coming from Futaba and this is just awesome. Man-I am so glad this worked.

dlktdr, 2022-09-27:
An I have a photo I can use now.. I had thus only tested from the bottom port

skunkwurk, 2022-09-27:
I noticed that in the headtracker doc, it only referenced the bottom port. I want to hide the BLE unit in the radio so I was exploring this option instead.  

BTW - I just switched my current mixer from TR11 to TR5 and it worked perfectly.  I don't have to touch anything to make this work except for the 2 settings and I'm good.

skunkwurk, 2022-09-27:
Last question, since I'm only using it for pan/tilt, will I get any latency benefit by reducing the channels sent to two?

dlktdr, 2022-09-27:
Not over Bluetooth, it's internally transmitting at 80Hz for the 8chs (So probably a lot faster than you had before over PPM) still. 
At 8ch PPM is only around 44hz. At 10 is even slower.

Just make sure that SBUS is set to 80Hz as well, so your not limiting it here.

skunkwurk, 2022-09-27:
Will do, thanks again!

closed 2022-09-27 by skunkwurk

dlktdr, 2022-09-27:
Added to the docs...

https://app.gitbook.com/o/-Md3r5nfSt9hl-HcRAzT/s/-Md5nSb70RWQSfl1nJPs/getting-started/wiring/option-2-wired/serial-bus-sbus/radio-master-tx16s-mk2

skunkwurk, 2022-09-27:
Pics before I closed the radio back up.  Looking forward to trying it this weekend.

[image]
[image]

gcmcnutt, 2022-10-16:
A nice looking install.  I'm about to do this to my mkII also -- thanks for leading the way!

skunkwurk, 2022-10-16:
> A nice looking install. I'm about to do this to my mkII also -- thanks for leading the way!

Thanks!  I've been using it a lot with zero issues.  It is a much better experience wireless.