headtracker/HeadTracker #133: Release 2.2 - Release Checklist

dlktdr, 2022-12-10:
Below is a list of things that need to be completed for the 2.2 release. The current branch for the 2.2 Release is under v2.2x

### Items still on the go that would be good to have for 2.2

- [ ] Gyro Calibration - Do on boot or in GUI? #130 
- [ ] Web App Crashing on Changing Gain #128 
- [ ] Qt6 Mac OS and Linux Builds? #125

### Items that need to be confirmed working

- [ ] PPM Output
- [ ] PPM Input
- [ ] SBUS Input
- [ ] SBUS Output
- [ ] Bluetooth Input
- [ ] Bluetooth Output
- [ ] CRSF input
- [ ] CRSF output
- [ ] Analog Input
- [ ] PWM Output
- [ ] Aux Functions

If your willing to help out and can test one of the above items, please leave a comment below. Or something I missed above. Please test all actions from the 2.2x branch. The changes will be picked into here as they are completed.

OG-ALviK, 2022-12-10:
I have tested SBUS in (SBUS coming from a PC through UART FTDI adapter) and SBUS out (to atx16s radio) and can confirm that they are working perfectly!
I've also tested 3x ADC inputs (connected my Car Sim pedals to the BLE module's Analong pins) and they were working perfectly!
I would love to test CRSF in and out but so far i don't see a way to have it working in & out simultaneously...

rickgode, 2022-12-21:
I tested the PPM output on 2.2 and it works great.  The board froze on me as well changing the gain, but I was able to change the channels and the min/max of the outputs and those settings seemed to take.

bryceartstew, 2023-01-04:
The current AUX functions have been working great for me. I'm beside myself with where I have gotten with my project up to this point and It relates directly to the AUX functions. If you have an updated version of AUX functions that relate to issue #135 I haven't tested that. I'd be happy to play with the CRSF in/out but haven't yet

dlktdr, 2023-01-04:
> The current AUX functions have been working great for me. I'm beside myself with where I have gotten with my project up to this point and It relates directly to the AUX functions. If you have an [...]

Awesome! Just to clarify you have tested this on this 2.2 version, builds from here - https://github.com/dlktdr/HeadTracker/actions/workflows/build-firmware.yml?query=branch%3Av2.2x

The #135 extended limits shouldn't affect anything on the AUX. This was for the Tilt/Roll/Pan limits.

Out of curiosity what Aux outputs are you using and if you need others or different ranges please make an issue. I didn't actually know too many that are using them, glad to hear some feedback.

bryceartstew, 2023-01-05:
> > The current AUX functions have been working great for me. I'm beside myself with where I have gotten with my project up to this point and It relates directly to the AUX functions. If you have an updated version of AUX functions that relate to issue #135 I haven't tested that. I'd be happy to play with the CRSF in/out but haven't yet
> 
> Awesome! Just to clarify you have tested this on this 2.2 version, builds from here - https://github.com/dlktdr/HeadTracker/actions/workflows/build-firmware.yml?query=branch%3Av2.2x **Edit** - Nevermind I see you were the one who mentioned the web live issue, must be here :)
> 
> The #135 extended limits shouldn't affect anything on the AUX. This was for the Tilt/Roll/Pan limits.
> 
> Out of curiosity what Aux outputs are you using and if you need others or different ranges please make an issue. I didn't actually know too many that are using them, glad to hear some feedback.

No, I did not test on the 2.2 version. I've never pushed a firmware update outside of the app. I'll do my first manual firmware push tomorrow and update you.

I am using A4-7 as aux. I'd like to add some switches as well and could easily use a0-4 as additional AUX channels. I'll add a new issue.

closed 2023-06-19 by dlktdr