headtracker/HeadTracker #138: Allow for additional AUX channels to be used

bryceartstew, 2023-01-05:
A way for users to select if they want 1-8 aux channels using A0-7 on the board.

dlktdr, 2023-01-05:
Do you mean extra Analog inputs? The AUX outputs put out the Raw Gyro, Accelerometer, etc. values to channels. 

I think extra analogs is a good idea, or a choice at least if they should be analog or PWM. True resource mapping would be a nice to have.

bryceartstew, 2023-01-05:
I like the idea of True resource mapping. Didn't know how to word it. I've been reading a lot today about I2C because I was thinking of how it could be nice to add two lcd circle screens for channel output display. I see A4(SDA), A5(SCL) are important for that function so I'd lose two AUX channels to make that happen. I'm using my setup with two boards. One attached to the head set for head tracking and a second one stationary mounted to my flight sticks and plugged in through the trainer port 1/8"