headtracker/HeadTracker #146: Re-programming a Receiver as a Transmitter Happymodel EP2

ewcook, 2023-05-01:
Unable to get EP2 with TX firmware to bind with Radiomaster RX PWM.

https://headtracker.gitbook.io/head-tracker/getting-started/wiring/elrs-setup

I tried both 3.1.2 and 3.0.0 on the Radiomaster receiver without any luck. I have also verified the binding phrase. 

Please let me know if there is another firmware available for the EP2 to run as a TX, or if there is a procedure I can follow to build my own. 

Thanks,
Ed

ewcook, 2023-05-04:
Wired rx-tx and tx-rx on the board. Updated the firmware and select CRFS output, but it still does not bind.

ewcook, 2023-05-22:
I confirmed I have Part inverted selected in the settings.

dlktdr, 2023-05-24:
I just confirmed what I have on mine, I have UART inverted checked as well. [image]

There are not many users that I know of doing this so I would like to help track down your issue and to help future users. 

Just released a 2.2-pre-release with firmware included. If you can flash this firmware with this GUI to be sure we are on the same page.
https://drive.google.com/file/d/1DVPoDp4ZXhoI9SxvCinXkAfCYj1iWv1L/view?usp=share_link

Could you post photos of the wiring. Most of my tests have been with the DTQ Head Tracker board, mainly because it has 5v easily available. Are you using this with a Nano 33 BLE?

What RM PWM receiver do you buy. I can order up one of them for testing. I'm looking forward to checking them out anyway. Been using CRSF to PWM boards currently... less soldering with a direct to PWM 😄

ewcook, 2023-06-08:
Here is my wiring 

[image]

I have a HappyModel EP2 receiver and Radiometer ER5C PWM Receiver. 

Thank you for all the help!

andrzejroszkowski, 2023-06-27:
There is new feature in v3.3 called AirPort, maybe it is worth to check if it will work in RX2RX scenario... I've asked a question in this PR: https://github.com/ExpressLRS/ExpressLRS/pull/1904

ewcook, 2023-07-14:
Decided to continue using my TX16S as the communication link

ewcook, 2023-07-14:
Decided to continue using my TX16S for a communication link

closed 2023-07-14 by ewcook