headtracker/HeadTracker #151: Re-center using bluetooth

haplm, 2023-06-24:
It would be lovely to have a button on the RC remote to re-center (I'm using it on car, and there's no way I could reach out to my goggles and press a button there).

So I have two options
1. Either run the recenter button on a wire all the way down to the RC controller, which isn't really practical
2. Or build a simple BLE button based on something like this https://www.dfrobot.com/product-1259.html

Would it be possible to add a feature that would use the bluetooth connection on the headtracker to receive input from such a button?

dlktdr, 2023-06-24:
If you have another Nano, DTQSYS or even an ([XIAO Sense board](https://www.digikey.ca/en/products/detail/seeed-technology-co-ltd/102010469/16652896), if running the development version) You can do this now. Connect them together with one set as Bluetooth Head and the (button one) as Remote, when you press the reset button on the Remote it will cause the Head board to re-center. XIAO Sense firmware is now a part of the main dev, look in the actions for a .uf2 file you can write.

I have one of these boards on the way to me as well https://www.digikey.ca/en/products/detail/seeed-technology-co-ltd/102010448/16652893, it can't do headtracking, no IMU but would make an awesome receiver board with very little code changes.

Both XIAO boards also supports a running off a single cell lipo, with charger too which is handy.

haplm, 2023-06-25:
I do have another nano, so I could use it, yes.

The smaller XIAO board looks awesome - I plan to buy a car remote that has a trainer port, so I could use it both as a receiver and center button. I'm looking forward to you possibly adding support for it.

Also thank you for the last paragraph. Very interesting, theoretically if I could find something really small that could be programmed, it would also be an option.

closed 2023-06-25 by haplm