headtracker/HeadTracker #153: Not Working on DX8 g2

Thomas-GT40, 2023-06-30:
Hi my DX8 g2 don't see the board when i connect it with mono jack cable..
Do u have any idea ?

dlktdr, 2023-06-30:
On Specktrum, it has to be plugged into the trainer port after power up.

In the GUI make sure you select some channels for pan and tilt. Bottom right of the page, probably 5 and 6 for a dx8.

Double check plugged into trainer and not the audio port too has also come up.

MotionSic made a video on the Dx12, should be pretty similar for you 

https://youtu.be/zpi_qeuuT8k

Thomas-GT40, 2023-06-30:
sadely it don't work, my dx8 detect it like 1 sec if i click on the reset button while it is connected, but that's all.
I've tried all the version of the software on your website, but nothing =(

Thomas-GT40, 2023-06-30:
Also, it worked on my old DX7s, but sadely i had to change for 8 channel... that's why the dx8 g2

dlktdr, 2023-06-30:
Maybe on the output/pwm page. Try playing with the number of channels that are output, frame length and sync values? Or possibly invert it (unlikely this one)

Wouldn't happen to have an oscilloscope would you? Could first of all make sure there is a signal there. Second could put your radio in student mode to see the specs of the signal then set the GUI to match.

Thomas-GT40, 2023-07-02:
Thanks for you answer, i don't have acces to any oscilloscope. i will try to play with the number of channel.
i'll keep you in touch. But would it be possible to make it work in bluetooth ?

Thomas-GT40, 2023-07-09:
Hi, i didn't succeed to make it work.. But i've discovered a very weird things. When i reset the arduino board while it is connected to the Spektrum DX8, the radio see the arduino for like 1 sec until the orange led start to blink.
I've made a little video of it : 
https://drive.google.com/file/d/1PXU2Bs-PkRqE7yq-Q3-OD1hrFjrpdWGm/view?usp=sharing 

I precise that i update the arduino with the latest version of the program (2.13).
Thanks for your help

dlktdr, 2023-07-09:
> But would it be possible to make it work in bluetooth ?

No unfortunately Bluetooth is only compatible with FrSky modules.

> I've made a little video of it :

Can't seem to watch it, no permission.

> I precise that i update the arduino with the latest version of the program (2.13).

The newest FW is v2.21, GUI 2.2. https://github.com/dlktdr/HeadTracker/releases

Do you have a multimeter, another way to verify you have a signal there and it's doing the right thing, measure from tip to the base of your 3.5mm jack. Should get apx 2.8 (average of all the pulses) the value isn't super important, then click invert ppm output in the GUI, it should switch to 3.3v minus that value. So apx, 0.5v. Normally for Specktrum you don't want inverted after your done.

It should do the same when plugged into the radio.

Thomas-GT40, 2023-07-09:
I'll be able to try with a multimeter next week-end. I hope it will work, else it is the radio.. I think should try with another one.

thanks

closed 2025-03-15 by dlktdr