headtracker/HeadTracker #154: ELRS Backpack wireless headtracker function support

xuhao1, 2023-07-23:
Hi there! Thanks for your work.
ELRS now added native Bluetooth support for headtracker, as shwon in https://github.com/ExpressLRS/ExpressLRS/pull/2060 I think it would be great to add the mode supported in this project so that we can directly use wireless headtracker with any ELRS transmitter system without any hack on hardware!

pkendall64, 2023-07-23:
ELRS is not using bluetooth, it uses MSP packet sent over ESP-now.

xuhao1, 2023-07-24:
@pkendall64 okay I find this. How about mounting a ELRS receiver on the head tracker module to enable backpack?

NineDayCC, 2023-08-01:
Hi, I'v been working on a headtracker project based on @dlktdr 's headtracker on esp32-C3. The main structure has been done. It can output ppm and subs signals, wireless connection through bluetooth receiver. 

But many features are waiting to be added like gui parameter configurator support, and some issues needed to be fixed (such as ppm output jitter）. 

Maybe you can take a look in my repository and see if we can make it better together(I'll upload some time later). I chose esp32-C3 because it is very cheap, so everyone can afford it. I hope I can contribute my effort to make community better!

dlktdr, 2023-08-01:
> Hi, I'v been working on a headtracker project based on @dlktdr 's headtracker on esp32-C3. The main structure has been done. It can output ppm and subs signals, wireless connection through bluetooth [...]

Hey, that sounds awesome, I really wanted to get this working on ESP, but haven't had the time to really put the effort in. It is a really nice chip and the extra processing power + Wifi is a big plus. Could use ESPnow directly to the ELRS too if we could get it running.

I would be glad to help out. Looks like your still using ZpehyrOS from the question you posted, post the code too when you can. 

For the PPM I was looking into using the RMT peripheral on ESP to generate the pulses. Same kind of jitter issue was found on NRF when using interrupts as well. Really only PPM, PWM and getting it compiling with West for the newest ESP support should be needed. If you have discord, send me a PM or join the #dev channel.

dlktdr, 2023-08-01:
The link to discord is, https://discord.gg/ux5hEaNSPQ

L1cardo, 2024-03-15:
How's this going on?
Is esp32 getting support soon?

ltwin8, 2024-07-12:
hi there, are there any updates on this?

ESP8285, standard IMU, thats all needed if i understand correctly?

like if one has an EP82 vrx backpapck already, can it be possible to add headtracker? or some system with esp+imu for internal fit to (digital) goggles