headtracker/HeadTracker #155: Unable to open bossac_mac

pguillory, 2023-08-06:
I have a Motionsic head tracker and a Mac. I downloaded the Headtracker v2.2 GUI and tried to flash the firmware using Firmware Loader. I get the error "Unable to open bossac_mac".

[image]

The log looks like this:

```
Loading local file /Users/pguillory/Downloads/BLE v2.21.bin
Using firmware from local file /Users/pguillory/Downloads/BLE v2.21.bin
Determining what type of file this is...
Firmware is for the Arduino Nano BLE 33 in bin format
Discovered a new port tty.usbmodem101
Wasn't able to set bootloader mode, trying first port found
Starting: bossac_mac --port=tty.usbmodem101 -e -w -R /Users/pguillory/Downloads/BLE v2.21.bin
Killed programmer process!
```

I'm not sure where to get bossac_mac, it's not a program on my system. I checked on Homebrew and there's a package called "bossa", however it installs a program called `bossac`, not `bossac_mac`.

Is there a step I missed?

dlktdr, 2023-08-06:
No nothing you missed, we weren't able to get bossac for mac ready to go before 2.2. It needed to be signed so there was some extra work involved. @gcmcnutt is working on this one.

[image] I put a note on the download page, but your not the first to miss it :)

If you have a copy, you should be able to just rename it to bossac_mac. The bossa version used for the Nano33BLE of bossac was customized by Arduino. It would probably be safer to use the bossac that comes with the ArduinoIDE. 

If you can flash it on another device you should be able to still configure on Mac.

Iragne, 2024-04-04:
FYI i used  /Users/XXX/Library/Arduino15/packages/arduino/tools/bossac/1.9.1-arduino2/bossac

Iragne, 2024-04-04:
Let me add more detail, used my arduino ide download the board driver associated to Nano 33 BLE rev2, validate that now Arduino ide was able to reconnized my board and use the following command

 bossac_mac --port=tty.usbmodem1201 -e -w -R ./BLE_SR2_v2_21_6876704.bin        <= this version is for the rev 2 , i found it in the discord

Iragne, 2024-04-05:
@dlktdr  happy to update doc if i found it